OpenXiangShan/XiangShan #2731: suffix missing in python code

linkedlist771, 2024-03-01:
#### Before start
PLEASE MAKE SURE you have done these: 
- [x] (Select what you have done like this)
- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question.
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest).
- [x] I have searched the previous issues and did not find anything relevant.
- [x] I have reviewed the commit messages from the relevant commit history.

#### Describe the bug
The python [script file’s](https://github.com/OpenXiangShan/XiangShan/tree/master/scripts) suffix is missing. 

#### Screenshots
None

#### Expected behavior
None

#### To Reproduce
None

#### Environment (please complete the following information):
 None

#### Additional context
Add any other context about the problem here.

eastonman, 2024-03-01:
Why does suffix matter?

linkedlist771, 2024-03-01:
It is a standard convention in Python programming that programmers are expected to follow.  Actually, this convention may not be so important in this C++ based project.

eastonman, 2024-03-01:
Ok, so I guess PR is welcome?

chenguokai, 2024-03-01:
PR is welcome :-)

poemonsense, 2024-03-03:
The script is used as a Linux executable script. Linux automatically reads the first line `#! /usr/bin/env python3` and uses Python to execute it.

closed 2024-03-03 by poemonsense