OpenXiangShan/XiangShan #2769: Questions about performing spec2006 tests with Xiangshan Simulator

Ergou-ren, 2024-03-11:
#### Before start
PLEASE MAKE SURE you have done these: 
- [x] (Select what you have done like this) 
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 
  - [x] I just followed the description in the documents. 
  - [x] I cannot find solutions in the documents. 
- [x] I have searched the previous issues and did not find anything relevant. 

#### Describe you problem. 
I successfully compiled NEMU and Xiangshan emulator, and I also successfully compiled spec2006 test program, and I successfully used NEMU to get information such as checkpoint. However, when I performed these checkpoints using Xiangshan Simulator, the result I got seemed to be not a correct one. I hope to get your help. 

#### Screenshots 
[image]
[image]
We can see that he executed 1 instruction, which doesn't seem to make much sense, but I don't know where I made a mistake.
[image]
But when I executed it with NEMU, he could print the correct result.
 

#### What did you do before 
Please describe step by step what did you do before you encounted the problem. 

#### Environment (please complete the following information): 
 - Operating System: Linux oberon 6.5.0-25-generic #25~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Feb 20 16:09:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
 - gcc version: gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0 Copyright (C) 2021 Free Software Foundation, Inc.
 - mill version: Mill Build Tool version 0.11.6
Java version: 11.0.22, vendor: Ubuntu, runtime: /usr/lib/jvm/java-11-openjdk-amd64
Default locale: zh_CN, platform encoding: UTF-8
OS name: "Linux", version: 6.5.0-25-generic, arch: amd6
 - java version: openjdk 11.0.22 2024-01-16
OpenJDK Runtime Environment (build 11.0.22+7-post-Ubuntu-0ubuntu222.04.1)
OpenJDK 64-Bit Server VM (build 11.0.22+7-post-Ubuntu-0ubuntu222.04.1, mixed mode, sharing)
 - XiangShan branch: [image]
 - XiangShan commit id:[image]
 - NEMU commit id: [image]
 - SPIKE commit id: [image]

#### Additional context 
nothing

poemonsense, 2024-03-11:
The latest NEMU cannot co-simulate with the legacy nanhu branch of XiangShan. 

If you are going to co-simulate nanhu XiangShan against NEMU, please try either 
- using the pre-compiled nemu so file in `ready-to-run` directory, or
- checkouting nemu to the similar time of nanhu branch of XiangShan (~Feb 2023)

Ergou-ren, 2024-03-15:
> The latest NEMU cannot co-simulate with the legacy nanhu branch of XiangShan.
>
> If you are going to co-simulate nanhu XiangShan against NEMU, please try either
>
> * using the pre-compiled nemu so file [...]

Thank you, I have succeeded according to your plan, but I don't know whether this is the expected output, it has been executed for 3 days, I can't confirm whether it is in a normal test.
I used the second plan you gave me.
[image]

poemonsense, 2024-03-15:
> Thank you, I have succeeded according to your plan, but I don't know whether this is the expected output, it has been executed for 3 days, I can't confirm whether it is in a normal test.

For large-scale programs, its length may be quite long such that it does not exit within days of simulation. For SPEC checkpoints, they may take years of RTL-simulation to complete. So I think you have already run it correctly.

Usually we run checkpoints for a fixed time/instruction interval. Using `-I 12345` will run 12345 instructions. Similarly, using `-C 12345` runs for 12345 clock cycles. You can set a suitable number when simulating xiangshan with the checkpoints.

poemonsense, 2024-03-15:
Since the original NEMU issue is resolved, I'm closing this issue now

closed 2024-03-15 by poemonsense