OpenXiangShan/XiangShan #2780: Assertion failed when running EMU on nanhu-G branch

DimanChauncey, 2024-03-12:
#### When running emu for hello workload on nanhu-G branch, Assertion failed.
```
root@localhost:~/nexus-am/apps/hello# time /root/XiangShan/build/emu --no-diff -i /root/nexus-am/apps/hello/build/hello-riscv64-xs.bin
Emu compiled at Mar 10 2024, 16:21:12
Using simulated 32768B flash
No valid flash bin path, use preset flash instead
The image is /root/nexus-am/apps/hello/build/hello-riscv64-xs.bin
Using simulated 8192MB RAM
Assertion failed at line 2569643.
Assertion failed at line 2569643.
The simulation stopped. There might be some assertion failed.
Core 0: ABORT at pc = 0x1150fafead
total guest instructions = 3
instrCnt = 3, cycleCnt = 8203, IPC = 0.000366
Seed=0 Guest cycle spent: 8206 (this will be different from cycleCnt if emu loads a snapshot)
Host time spent: 39490ms
Assertion failed: Error L3: meta still conflict when probe helper finish! mshrId:  0
    at MSHR.scala:938 assert(RegNext(!clients_meta_busy, true.B),
Assertion failed: Error L3: meta still conflict when probe helper finish! mshrId:  0
    at MSHR.scala:938 assert(RegNext(!clients_meta_busy, true.B),
```

#### Screenshots
[image]

I created the environment according to the official website documentation, and both make sim verilog and make emu were successful

#### Environment
 - Operating System: [Ubuntu 20.04]
 - gcc version: [gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.2) ]
 - mill version: [0.9.8]
 - java version: [openjdk 11.0.22 2024-01-16]
 - XiangShan branch: [nanhu-G newest]

poemonsense, 2024-03-13:
Which verilator verison are you using? It's a known issue Verilator v5 leads to these assertions on nanhu. I didn't have time to locate the issue and fix it.

DimanChauncey, 2024-03-13:
感谢。确实是v5版本，Verilator 5.020 2024-01-01 rev v5.020，求一个可用的verilator版本

cebarobot, 2024-03-13:
Please consider to use verilator v4.218.

poemonsense, 2024-03-13:
> Thank you. It is indeed v5 version[Verilator 5.020 2024-01-01 rev v5.020], could you tell me a usable version?

As suggested, please try v4.218.

We still don't know whether the assertions with v5 are due to Verilator bugs or RTL bugs. However, v5 works well for the master branch of XiangShan.

@wakafa1 anyone have free time and interested to analyze the waveform of nanhu? Previously I dumped the waveform but didn't have time to take a look. If someone knows huancun well, maybe it takes only minutes to find the real issues.

poemonsense, 2024-03-13:
> Please consider to use verilator v4.218.

It's worth noting v4.218 do have bugs. The bug simply does not manifest on XiangShan RTL. But it do have bugs.

poemonsense, 2024-03-13:
> As suggested, please try v4.218.
>
> We still don't know whether the assertions with v5 are due to Verilator bugs or RTL bugs. However, v5 works well for the master branch of XiangShan.
>
> @wakafa1 [...]

Is it because of the clock gating cell?

wakafa1, 2024-03-13:
> Is it because of the clock gating cell?

Most likely. I can take a look.

poemonsense, 2024-03-13:
Thanks for reporting the issue. We have fixed the clock gating cell for both nanhu and nanhu-G branches. They are tested to work well with Verilatro v5.

Please have a try. Verilator v5 is expected to have better compilation speed and simulation speed.

closed 2024-03-13 by poemonsense