OpenXiangShan/XiangShan #2813: the logic of tlbBundle privilege in dmode

yelbuod, 2024-03-26:
#### Before start
PLEASE MAKE SURE you have done these: 
- [x] (Select what you have done like this) 
- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 
- [x] I have searched the previous issues and did not find anything relevant. 
- [x] I have searched the previous discussions and did not find anything relevant. 
- [x] I have reviewed the commit messages from the relevant commit history. 

#### Describe the question 
A clear and concise description of your question.
I looked at the description of dcsr in the riscv debug specification, and the `mprven `bit field of  is described below:
> 0: MPRV in mstatus is ignored in Debug Mode. 1: MPRV in mstatus takes effect in Debug Mode.
But the logic of `tlbBundle.priv.dmode` below seems to be the opposite of what is defined in the spec
https://github.com/OpenXiangShan/XiangShan/blob/4176c33937a5547113be7a0c2411a74f09cc2dfd/src/main/scala/xiangshan/backend/fu/CSR.scala#L902
when debugMode is true and mprven is set to 1, mprv in mstatus seems to be ignored. If there is any misunderstanding on my part, please feel free to correct me and provide guidance.

wissygh, 2024-03-27:
This has been fixed in #2740

yelbuod, 2024-03-27:
Sorry I didn't check the PR

closed 2024-03-27 by yelbuod

Tang-Haojin, 2024-03-27:
We will merge this into master in near future.