OpenXiangShan/XiangShan #2858: 香山环境如何使用Verilator 的profile 功能

guyandeli, 2024-04-08:
我按照Verilator 手册添加profile 选项（--prof-cfuncs），无法产生gmon.out 分析报告（我跑的coremark 可以正常结束 ），请问在香山环境中如何使用profile 功能。

#### Before start
PLEASE MAKE SURE you have done these: 
- [x] (Select what you have done like this) 
- [ ] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 
- [ ] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 
- [ ] I have searched the previous issues and did not find anything relevant. 
- [ ] I have searched the previous discussions and did not find anything relevant. 
- [ ] I have reviewed the commit messages from the relevant commit history. 

#### Describe the question 
A clear and concise description of your question.

cebarobot, 2024-04-08:
首先，请您按照 Issue 模板的指示完成提问前步骤。
First, Please follow issue template to finish steps before asking questions.

其次，请介绍更多细节，我们尚不清楚您在哪里添加的 `--prof-cfuncs`。
Second, please introduce more details. We still do not know where you add `--prof-cfuncs`.

guyandeli, 2024-04-09:
我是在difftest/verilator.mk 中添加 --prof-cfuncs

poemonsense, 2024-04-09:
Please try CONFIG=MinimalConfig. It is a smaller config of XiangShan. It should produce the gmon.out correctly.

We don't know why the DefaultConfig cannot produce gmon.out. It could be most likely due to Verilator issues.

guyandeli, 2024-04-12:
I tried it here and it turns out that gmon.out can only be generated in MinimalConfig mode, but not in default mode. But I have to use the default mode here. Can you help me see how to generate the profile file in the default mode? Is it because the code size in the default mode is too large and cannot be generated?

poemonsense, 2024-04-12:
As I said before, we don't know why the DefaultConfig cannot produce gmon.out. It could be most likely due to Verilator issues.

Please reach out to Verilator for help. https://github.com/verilator/verilator/issues

poemonsense, 2024-04-12:
It's worth noting DefaultConfig and MinimalConfig does not affect the simulation process. So we expect no differences for the workflow to profile the simulation.

We did notice that DefaultConfig cannot produce gmon.out before. However, we don't have enough resources to figure out the reason. We are happy if you would like to identify the real issues here.

cebarobot, 2024-09-14:
We see that you haven't responded to this issue for a long time. If you have any other questions, please feel free to open the issue again or file a new one.

我们看到您已经很久没有回复本 issue。如果您有其他疑问，欢迎再次打开 issue 或提交新的 issue。

closed 2024-09-14 by cebarobot