OpenXiangShan/XiangShan #2886: Using XiangShan to reproduce the example code in RISC-V Architecture Programming and Practice.

Danling-droid, 2024-04-16:
#### Before start
PLEASE MAKE SURE you have done these: 
- [x] (Select what you have done like this) 
- [x] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 
- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 
- [x] I have searched the previous issues and did not find anything relevant. 
- [x] I have searched the previous discussions and did not find anything relevant. 
- [x] I have reviewed the commit messages from the relevant commit history. 

#### Describe the question 
I want to run the example code from `https://github.com/runninglinuxkernel/riscv_programming_practice` on XiangShan and use AM to regenerate the code to suit XiangShan's workload. Can you give me some advice?

I made modifications according to the compilation and linking methods of the example code myself, but it got stuck at the following location during execution：
```shell
DL@zz:~/XiangShan$ ./build/simv +workload=../nexus-am/apps/benos/benos_payload.bin +no-diff
Chronologic VCS simulator copyright 1991-2018
Contains Synopsys proprietary information.
Compiler version O-2018.09-SP2_Full64; Runtime version O-2018.09-SP2_Full64;  Apr 16 09:53 2024
ram image:../nexus-am/apps/benos/benos_payload.bin
disable diff-test
simv compiled at Apr  8 2024, 11:49:53
Using simulated 8192MB RAM
The image is ../nexus-am/apps/benos/benos_payload.bin
Using simulated 32768B flash

```

cebarobot, 2024-04-16:
Please give more infomation of your workload. 

By the way, have you tried to run your workload on NEMU?

poemonsense, 2024-04-16:
Recommend using Verilator and running the simulation with `--dump-commit-trace` to see the commit logs.

Danling-droid, 2024-04-18:
Sorry for not replying in time.

XiangShan commit ID: 1ce20194b955adf5a2f09592c40ff9e2dbaf8837

Also, the results obtained using the source code example in chapter_2 of the book are as follows:
```shell
DL@zz:~/XiangShan$ ./build/simv +workload=../nexus-am/apps/test/benos_payload.bin
Chronologic VCS simulator copyright 1991-2018
Contains Synopsys proprietary information.
Compiler version O-2018.09-SP2_Full64; Runtime version O-2018.09-SP2_Full64;  Apr 18 16:47 2024
ram image:../nexus-am/apps/test/benos_payload.bin
simv compiled at Apr  8 2024, 11:49:53
Using simulated 8192MB RAM
The image is ../nexus-am/apps/test/benos_payload.bin
Using simulated 32768B flash
The reference model is /home/DL/NEMU/build/riscv64-nemu-interpreter-so
The first instruction of core 0 has commited. Difftest enabled.
[src/memory/paddr.c:171,paddr_read] isa pmp check failed
ERROR: invalid mem read from paddr 0x0000000000000000, NEMU raise access exception
ERROR: invalid mem read from paddr 0x0000000000000000, NEMU raise access exception
ERROR: invalid mem read from paddr 0x0000000000000000, NEMU raise access exception
ERROR: invalid mem read from paddr 0x0000000000000000, NEMU raise access exception
```

So what should I do to use the source code in the book to generate a workload using AM?

poemonsense, 2024-04-18:
Your workload is accessing address 0x0 and the CPU keeps raising exceptions on this address.

The most likely reason is that, your workload triggers some exception while the exception handler address (`mtvec`) is not set, which would thus be the default value of 0. To handle the exception, CPU goes to 0x0. However, this address is illegal, and therefore, the CPU raises an access fault exception again and again.

As we have suggested, please first confirm that your workload is correct in terms of its semantics. You can either 1) test your workload on NEMU and see how your workload/program/instruction sequences are executed, or 2) enable commit trace on XiangShan, by using `--dump-commit-trace` when using the Verilator simulator.

Danling-droid, 2024-04-22:
Hello, here are the commit records captured using Verilator:
```shell
[11] commit pc 000000008000006a inst 18079073 wen 0 dst 00 data 0000000000000000 idx 02b
[12] commit pc 000000008000006e inst 30200073 wen 0 dst 00 data 0000000000000000 idx 02c
[13] exception pc 0000000080200000 inst 00001117 cause 0000000000000001
^[[1;34m[src/memory/paddr.c:171,paddr_read] isa pmp check failed^[[0m
[14] exception pc 0000000000000000 inst 00001117 cause 0000000000000001
ERROR: invalid mem read from paddr 0x0000000000000000, NEMU raise access exception
[15] exception pc 0000000000000000 inst 00001117 cause 0000000000000001
ERROR: invalid mem read from paddr 0x0000000000000000, NEMU raise access exception
```
Additionally, here's a portion of the disassembled file corresponding to the workload being used:
```shell
80000068: 6782 ld a5,0(sp)
8000006a: 18079073 csrw satp,a5
8000006e: 30200073 mret
80000072: 0001 nop
80000074: 7462 ld s0,56(sp)
80000076: 6121 add sp,sp,64
80000078: 8082 ret

Disassembly of section .load:

0000000080200000 <load_bin>:
80200000: 00001117 .word 0x00001117
80200004: 00010113 .word 0x00010113
```
But why is there an exception occurring at address 0x80200000?

poemonsense, 2024-04-22:
Please list or upload full commit logs here for us to verify the reason.

However, this issue is mostly unlikely a bug in XiangShan. We request you again to read the RISC-V specs and understand the semantics of your program before opening bug reports. Please try your best first to verify how your program should execute on the RISC-V architecture.

cebarobot, 2024-09-14:
We see that you haven't responded to this issue for a long time. If you have any other questions, please feel free to open the issue again or file a new one.

我们看到您已经很久没有回复本 issue。如果您有其他疑问，欢迎再次打开 issue 或提交新的 issue。

closed 2024-09-14 by cebarobot