OpenXiangShan/XiangShan #2903: How are the uncacheable and non-indempotent stores are handled?

XubsXubs, 2024-04-19:
Hi, to the developers of XiangShan,

My question is as above, I have seen there is a module named PATH/dcache/Uncache.scala, which mentioned uncacheable load and stores, and this is instantiated in the backend/memblock.scala as show here:

linjuanZ, 2024-04-26:
The code you quote means that if this request is a load, or a store that is not allowed to be outstanding, LSU should wait for response from uncache until the next request is accepted.

XubsXubs, 2024-04-29:
I read on the document that, for MMIO ld, for example, it is maintained in the load queue until it reaches the head of ROB, then it would be sent to the downwards; same seems to be for the MMIO store, that needs to wait for becoming the head of ROB, then can be send to downwards. 

What is the connection here, are the Uncache relevant here?

linjuanZ, 2024-05-13:
> I read on the document that, for MMIO ld, for example, it is maintained in the load queue until it reaches the head of ROB, then it would be sent to the downwards; same seems to be for the MMIO [...]

Uncache is a module that receives mmio loads/stores from LSQ and transforms the requests to TileLink messages.

closed 2024-06-05 by linjuanZ

XubsXubs, 2024-06-05:
Make sense, thanks!