OpenXiangShan/XiangShan #2934: Is there any interface to flush L2 Cache data

Phantom1003, 2024-04-28:
Hi, I'm currently doing some benchmarking on XiangShan. I'd like to inquire if the core has the capability of actively invalidating the L2 cache through instructions or control registers.

Thanks.

Tang-Haojin, 2024-05-08:
Sorry for the late reply. Currently, there is no such instruction yet.

LMiaoH, 2024-05-27:
In the Cache Controller of nanhu core, can the CMD_CMO_INV, CMD_CMO_CLEAN, CMD_CMO_FLUSH commands accomplish this operation? Or, in other words, can the Cache Controller be used to flush dirty data to memory? If so, how should it be done?

LMiaoH, 2024-05-27:
> Hi, I'm currently doing some benchmarking on XiangShan. I'd like to inquire if the core has the capability of actively invalidating the L2 cache through instructions or control registers.
> 
> 
In the Cache Controller of nanhu core, can the CMD_CMO_INV, CMD_CMO_CLEAN, CMD_CMO_FLUSH commands accomplish this operation? Or, in other words, can the Cache Controller be used to flush dirty data to memory? If so, how should it be done?