OpenXiangShan/XiangShan #2989: Compile Exception while using "make emu EMU_THREADS=8 MFC=1 CONFIG=KunminghuV2Config",

Jeremy-Jia, 2024-05-15:
#### Before start
PLEASE MAKE SURE you have done these: 
- [x] (Select what you have done like this) 
- [ ] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 
- [ ] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 
- [ ] I have searched the previous issues and did not find anything relevant. 
- [ ] I have searched the previous discussions and did not find anything relevant. 
- [ ] I have reviewed the commit messages from the relevant commit history. 

#### Describe the question 
A clear and concise description of your question.

### Compile Exception while using "make emu EMU_THREADS=8 MFC=1 CONFIG=KunminghuV2Config", the error message is:
        Exception in thread "main" java.util.NoSuchElementException: head of empty list
        at scala.collection.immutable.Nil$.head(List.scala:662)
        at scala.collection.immutable.Nil$.head(List.scala:661)
        at top.BusPerfMonitorImp.<init>(BusPerfMonitor.scala:105)
        at top.BusPerfMonitor.$anonfun$module$1(BusPerfMonitor.scala:30)
        at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
        at top.BusPerfMonitor.module$lzycompute(BusPerfMonitor.scala:30)
        at top.BusPerfMonitor.module(BusPerfMonitor.scala:30)
        at top.BusPerfMonitor.module(BusPerfMonitor.scala:28)
        at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$15(LazyModule.scala:335)
        at chisel3.Module$.evaluate(Module.scala:88)
        at chisel3.Module$.do_apply(Module.scala:35)
        at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$14(LazyModule.scala:335)
        at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
        at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$13(LazyModule.scala:335)
        at scala.Option.getOrElse(Option.scala:201)
        at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$1(LazyModule.scala:333)
        at scala.collection.immutable.List.flatMap(List.scala:293)
        at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate(LazyModule.scala:309)
        at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate$(LazyModule.scala:306)
        at freechips.rocketchip.diplomacy.LazyModuleImp.instantiate(LazyModule.scala:396)
        at freechips.rocketchip.diplomacy.LazyModuleImp.$anonfun$x$22$1(LazyModule.scala:398)
        at chisel3.internal.plugin.package$.autoNameRecursivelyProduct(package.scala:48)
        at freechips.rocketchip.diplomacy.LazyModuleImp.<init>(LazyModule.scala:398)
        at xiangshan.L2Top$L2TopImp.<init>(L2Top.scala:147)
        at xiangshan.L2Top.$anonfun$module$1(L2Top.scala:244)
        at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
        at xiangshan.L2Top.module$lzycompute(L2Top.scala:244)
        at xiangshan.L2Top.module(L2Top.scala:244)
        at xiangshan.L2Top.module(L2Top.scala:56)
        at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$15(LazyModule.scala:335)
        at chisel3.Module$.evaluate(Module.scala:88)
        at chisel3.Module$.do_apply(Module.scala:35)
        at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$14(LazyModule.scala:335)
        at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
        at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$13(LazyModule.scala:335)
        at scala.Option.getOrElse(Option.scala:201)
        at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$1(LazyModule.scala:333)
        at scala.collection.immutable.List.flatMap(List.scala:293)
        at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate(LazyModule.scala:309)
        at freechips.rocketchip.diplomacy.LazyModuleImpLike.instantiate$(LazyModule.scala:306)
        at freechips.rocketchip.diplomacy.LazyModuleImp.instantiate(LazyModule.scala:396)
        at freechips.rocketchip.diplomacy.LazyModuleImp.$anonfun$x$22$1(LazyModule.scala:398)
        at chisel3.internal.plugin.package$.autoNameRecursivelyProduct(package.scala:48)
        at freechips.rocketchip.diplomacy.LazyModuleImp.<init>(LazyModule.scala:398)
        at xiangshan.XSTile$XSTileImp.<init>(XSTile.scala:108)
        at xiangshan.XSTile.$anonfun$module$1(XSTile.scala:182)
        at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
        at xiangshan.XSTile.module$lzycompute(XSTile.scala:182)
        at xiangshan.XSTile.module(XSTile.scala:182)
        at xiangshan.XSTile.module(XSTile.scala:33)
        at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$15(LazyModule.scala:335)
        at chisel3.Module$.evaluate(Module.scala:88)
        at chisel3.Module$.do_apply(Module.scala:35)
        at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$14(LazyModule.scala:335)
        at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
        at freechips.rocketchip.diplomacy.LazyModuleImpLike.$anonfun$instantiate$13(LazyModule.scala:335)
[... 66 lines cut ...]
        at top.Generator$.execute(Generator.scala:34)
        at top.SimTop$.delayedEndpoint$top$SimTop$1(SimTop.scala:111)
        at top.SimTop$delayedInit$body.apply(SimTop.scala:97)
        at scala.Function0.apply$mcV$sp(Function0.scala:42)
        at scala.Function0.apply$mcV$sp$(Function0.scala:42)
        at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:17)
        at scala.App.$anonfun$main$1(App.scala:98)
        at scala.App.$anonfun$main$1$adapted(App.scala:98)
        at scala.collection.IterableOnceOps.foreach(IterableOnce.scala:575)
        at scala.collection.IterableOnceOps.foreach$(IterableOnce.scala:573)
        at scala.collection.AbstractIterable.foreach(Iterable.scala:933)
        at scala.App.main(App.scala:98)
        at scala.App.main$(App.scala:96)
        at top.SimTop$.main(SimTop.scala:97)
        at top.SimTop.main(SimTop.scala)
1 targets failed
xiangshan[chisel].test.runMain subprocess failed
make: *** [Makefile:162: build/rtl/SimTop.v] Error 1

### The commit log: 
commit 5961467fb71810763d53d8d244fbcdfa2b81fa81 (HEAD -> master, origin/master, origin/HEAD)
Author: Xiaokun-Pei <peixiaokun23z@ict.ac.cn>
Date:   Wed May 15 14:23:12 2024 +0800

    HPTW, RVH: add the flush function (#2982)

cebarobot, 2024-05-15:
Thank you for your report. This issue has been reproduced.

We will fix it latter.

cebarobot, 2024-05-15:
Sorry, this is not a bug.

`KunminghuV2Config` is only used for generating verilog of XiangShan with CHI and cannot be used for building emu.

You could:
- Use `make verilog EMU_THREADS=8 MFC=1 CONFIG=KunminghuV2Config` to generate verilog for `KunminghuV2Config` but not build emu.
- Use `make verilog EMU_THREADS=8 MFC=1` to build emu for default config, which is also Kunminhu V2.

Jeremy-Jia, 2024-05-16:
Thank you for your reply. Please note that in the XiangShan Makefile:

  IMAGE ?= temp
  CONFIG ?= DefaultConfig
  NUM_CORES ?= 1
  MFC ?= 1

The default CONFIG is set to DefaultConfig. When I execute "make emu EMU_THREADS=8 MFC=1", it first generates the Verilog for DefaultConfig. Consequently, the emulator is combined with the DefaultConfig, not the KunminghuV2Config. Should I run "make verilog EMU_THREADS=8 MFC=1 CONFIG=KunminghuV2Config" first to generate the KunminghuV2Config, so that the Makefile will skip the Verilog generation step?

Jeremy-Jia, 2024-05-16:
Thank you for your reply. Please note that in the XiangShan Makefile:

```Makefile
IMAGE ?= temp
CONFIG ?= DefaultConfig
NUM_CORES ?= 1
MFC ?= 1
```
The default configuration is DefaultConfig. When I run the command “make emu EMU_THREADS=8 MFC=1”, it initially generates the Verilog code for DefaultConfig, which lacks CHI features. As a result, the emulator uses the Verilog code for DefaultConfig instead of KunminghuV2Config.

And, if I first generate the Verilog code for KunminghuV2Config using “make verilog EMU_THREADS=8 MFC=1 CONFIG=KunminghuV2Config”, this action will be overridden when I subsequently execute “make emu EMU_THREADS=8 MFC=1”.

cebarobot, 2024-05-16:
My previous response might have caused some misunderstanding. Please allow me to clarify:

You could run either `make verilog EMU_THREADS=8 MFC=1 CONFIG=KunminghuV2Config` or `make emu EMU_THREADS=8 MFC=1`, but not run them sequentially.

At present, our emu infrastructure are not ready for CHI, which means you cannot build emu for XiangShan with CHI (`KunminghuV2Config`) by any means. The only workable output of `KunminghuV2Config` is verilog.

Jeremy-Jia, 2024-05-16:
“At present, our emu infrastructure are not ready for CHI, which means you cannot build emu for XiangShan with CHI (KunminghuV2Config) by any means. The only workable output of KunminghuV2Config is verilog.”

Got it.

cebarobot, 2024-09-14:
We see that you haven't responded to this issue for a long time. If you have any other questions, please feel free to open the issue again or file a new one.

我们看到您已经很久没有回复本 issue。如果您有其他疑问，欢迎再次打开 issue 或提交新的 issue。

closed 2024-09-14 by cebarobot