OpenXiangShan/XiangShan #3057: Assertion failed     at UserYanker.scala:63 assert (!out.r.valid || r_valid) // Q must be ready faster than the response

Jeremy-Jia, 2024-06-11:
#### Before start
PLEASE MAKE SURE you have done these: 
- [x] (Select what you have done like this) 
- [ ] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 
  - [ ] I just followed the description in the documents. 
  - [ ] I cannot find solutions in the documents. 
- [ ] I have searched the previous issues and did not find anything relevant. 

#### Describe you problem. 
A clear and concise description of what the problem is. 

#### Screenshots 
If applicable, add screenshots to help explain your problem. 

#### What did you do before 
Please describe step by step what did you do before you encounted the problem. 

#### Environment (please complete the following information): 
 - Operating System: [e.g. Ubuntu 20.04]
 - gcc version: [paste first line of the output of `gcc --version`  if applicable]
 - mill version: [paste first line of the output of `mill -i --version`  if applicable]
 - java version: [paste first line of the output of `java --version`  if applicable]
 - XiangShan branch: [paste branch name here if applicable]
 - XiangShan commit id: [paste commit id here if applicable]
 - NEMU commit id: [paste commit id here if applicable]
 - SPIKE commit id: [paste commit id here if applicable]

#### Additional context 
Add any other context about the problem here.

错误信息：
./build/emu -i ../nexus-am/apps/hello/build/hello-riscv64-xs.bin --no-dif --du
mp-wave
Emu compiled at Jun 11 2024, 23:45:04
Using simulated 32768B flash
No valid flash bin path, use preset flash instead
The image is ../nexus-am/apps/hello/build/hello-riscv64-xs.bin
Using simulated 8192MB RAM
Assertion failed at line 26639.
The simulation stopped. There might be some assertion failed.
Core 0: ABORT at pc = 0x2ad67c688d
total guest instructions = 0
instrCnt = 0, cycleCnt = 544, IPC = 0.000000
Core 1: ABORT at pc = 0xfffffff35a1e481d
total guest instructions = 0
instrCnt = 0, cycleCnt = 544, IPC = 0.000000
Core 2: ABORT at pc = 0x3c5695d5c1
total guest instructions = 0
instrCnt = 0, cycleCnt = 544, IPC = 0.000000
Core 3: ABORT at pc = 0x1b101e5ff8
total guest instructions = 0
instrCnt = 0, cycleCnt = 544, IPC = 0.000000
Seed=0 Guest cycle spent: 547 (this will be different from cycleCnt if emu loads a snapshot)
Host time spent: 1854ms
Assertion failed
    at UserYanker.scala:63 assert (!out.r.valid || r_valid) // Q must be ready faster than the response

cebarobot, 2024-06-11:
MinimalConfig is just a config fore single core smoke test and is not well tested. I don't think minimalconfig could work properly with 4 cores. Please consider to use DefaultConfig for multiple cores.

MinimalConfig 仅用于单核冒烟测试，并没有经过细致地验证。我不认为它能够在4核配置下正常运行。对于4核场景，请考虑使用 DefaultConfig。

cebarobot, 2024-09-14:
We see that you haven't responded to this issue for a long time. If you have any other questions, please feel free to open the issue again or file a new one.

我们看到您已经很久没有回复本 issue。如果您有其他疑问，欢迎再次打开 issue 或提交新的 issue。

closed 2024-09-14 by cebarobot