OpenXiangShan/XiangShan #3072: make verilog NUM_CORES=4 出错

meiqin0, 2024-06-14:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

**make verilog NUM_CORES=4 出错：**

### What did you do before

**make verilog NUM_CORES=4 出错：**
错误1：
[warn] There were 28 warning(s) during hardware elaboration.
Exception in thread "main" java.lang.OutOfMemoryError: Java heap space
	at scala.collection.mutable.ArrayBuffer$.scala$collection$mutable$ArrayBuffer$$ensureSize(ArrayBuffer.scala:332)
	at scala.collection.mutable.ArrayBuffer.ensureAdditionalSize(ArrayBuffer.scala:77)
	at scala.collection.mutable.ArrayBuffer.addOne(ArrayBuffer.scala:146)
	at scala.collection.mutable.ArrayBuffer.addOne(ArrayBuffer.scala:43)
	at scala.collection.mutable.Buffer.append(Buffer.scala:43)
	at scala.collection.mutable.Buffer.append$(Buffer.scala:43)
	at scala.collection.mutable.AbstractBuffer.append(Buffer.scala:232)
	at firrtl.ir.Block.mapStmt(IR.scala:653)
	at firrtl.Mappers$StmtMagnet$$anon$3.map(Mappers.scala:31)
	at firrtl.Mappers$StmtMap$.map$extension(Mappers.scala:48)
	at firrtl.passes.CInferTypes$.infer_types_s$2(InferTypes.scala:149)
	at firrtl.passes.CInferTypes$.$anonfun$run$39(InferTypes.scala:159)
	at firrtl.passes.CInferTypes$$$Lambda$94647/0x0000000806b80040.apply(Unknown Source)
	at firrtl.ir.Module.mapStmt(IR.scala:1198)
	at firrtl.Mappers$ModuleMagnet$$anon$14.map(Mappers.scala:108)
	at firrtl.Mappers$ModuleMap$.map$extension(Mappers.scala:121)
	at firrtl.passes.CInferTypes$.infer_types$2(InferTypes.scala:159)
	at firrtl.passes.CInferTypes$.$anonfun$run$41(InferTypes.scala:162)
	at firrtl.passes.CInferTypes$$$Lambda$94644/0x0000000806c2b040.apply(Unknown Source)
	at scala.collection.immutable.List.map(List.scala:250)
	at scala.collection.immutable.List.map(List.scala:79)
	at firrtl.passes.CInferTypes$.run(InferTypes.scala:162)
	at firrtl.passes.Pass.execute(Pass.scala:14)
	at firrtl.passes.Pass.execute$(Pass.scala:14)
	at firrtl.passes.CInferTypes$.execute(InferTypes.scala:100)
	at firrtl.Transform.transform(Compiler.scala:280)
	at firrtl.Transform.transform$(Compiler.scala:280)
	at firrtl.passes.CInferTypes$.transform(InferTypes.scala:100)
	at firrtl.stage.transforms.ExpandPrepares.execute(ExpandPrepares.scala:19)
	at firrtl.Transform.transform(Compiler.scala:280)
	at firrtl.Transform.transform$(Compiler.scala:280)
	at firrtl.stage.transforms.ExpandPrepares.transform(ExpandPrepares.scala:7)
1 targets failed
xiangshan[chisel3].test.runMain subprocess failed
make: *** [build/rtl/SimTop.v] Error 1

**之后修改了JVM的Xmx，大小为256G， 不再报此错误， 但出现了新错误：**
错误2：
FreeList: UncacheBuffer freelist, size 20
StoreQueue: size:64
Constantin initRead: LFSTEnable = 1
Constantin initRead: LFSTEnable = 1
Constantin initRead: LFSTEnable = 1
Constantin initRead: ForceWriteUpper_0 = 60
Constantin initRead: ForceWriteLower_0 = 55
Constantin initRead: StoreBufferThreshold_0 = 7
Constantin initRead: StoreBufferBase_0 = 4
Exception in thread "main" java.lang.NoClassDefFoundError: utility/ValidIODelay$
	at xiangshan.backend.MemBlockImp.$anonfun$new$87(MemBlock.scala:503)
	at chisel3.experimental.prefix$.apply(prefix.scala:50)
	at xiangshan.backend.MemBlockImp.$anonfun$new$86(MemBlock.scala:503)
	at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:33)
	at xiangshan.backend.MemBlockImp.$anonfun$new$85(MemBlock.scala:503)
	at scala.Option.foreach(Option.scala:437)
[... 139 lines cut ...]
	... 141 more
1 targets failed
xiangshan[chisel3].test.runMain subprocess failed
make: *** [build/rtl/SimTop.v] Error 1

**但是utility.ValidIODelay确实定义了**

### Environment

- XiangShan branch: master
- XiangShan commit id: commit 5794a5afda0002ad0014736a2b81824aab0616a0
- NEMU commit id: dont care, but commit 01e78348a826a835d77af6dff7a69eaefa27fecb
- SPIKE commit id:  dont care
- Operating System: Centos7.9
- gcc version: 4.8.5
- mill version: 0.11.7
- java version: 11.0.22

### Additional context

**不知道是哪里的问题？**

**java 11/17都尝试过；物理内存512G**

**应该不是OS的问题，因为南湖版本编译单核、双核、四核都未出现此情况**

Tang-Haojin, 2024-06-14:
Have you tried `make verilog NUM_CORES=1` in your environment?

cebarobot, 2024-06-14:
考虑到您在使用的是 CentOS 7.9，感觉还是环境配置的有些问题。如果可能，我们还是推荐使用 Ubuntu 22.04。

Considering you are using CentOS 7.9, It probably is environment problem. If possible, we recommand you to use Ubuntu 22.04.

能否请您提供下面这些命令的运行结果，来辅助我们判断环境问题：
Could you please provide output of the below commands:
```
mill -i -v
java --version
echo $JAVA_HOME
```

meiqin0, 2024-06-14:
> Have you tried `make verilog NUM_CORES=1` in your environment?

尝试了，生成单核rtl是可以正常生成的，没有错误

meiqin0, 2024-06-14:
> 考虑到您在使用的是 CentOS 7.9，感觉还是环境配置的有些问题。如果可能，我们还是推荐使用 Ubuntu 22.04。
> 
> Considering you are using CentOS 7.9, it is probably a environment problem. If possible, we recommand you to use Ubuntu 22.04.
> 
> 能否请您提供下面这些命令的运行结果，来辅助我们判断环境问题： Could you please provide output of the below commands:
> 
> 
> mill -i --version
```
Mill Build Tool version 0.11.7
Java version: 11.0.22, vendor: Oracle Corporation, runtime: /home/jdk-11.0.22
Default locale: en_US, platform encoding: UTF-8
OS name: "Linux", version: 3.10.0-1160.el7.x86_64, arch: amd64
```
> java --version
```
java 11.0.22 2024-01-16 LTS
Java(TM) SE Runtime Environment 18.9 (build 11.0.22+9-LTS-219)
Java HotSpot(TM) 64-Bit Server VM 18.9 (build 11.0.22+9-LTS-219, mixed mode)
```
> echo $JAVA_HOME
```
/home/jdk-11.0.22
```
> ```

cebarobot, 2024-06-14:
It's really strange... Have you tried to `make clean` or re-clone the XiangShan project?

这就很怪了。您尝试过 `make clean` 或者重新 clone 香山项目吗？

meiqin0, 2024-06-14:
> It's really strange... Have you tried to `make clean` or re-clone the XiangShan project?
> 
> 这就很怪了。您尝试过 `make clean` 或者重新 clone 香山项目吗？

尝试过make clean && rm -rf out ，也尝试过重新 clone代码。我刚才尝试，将jvm参数改成-Xmx200G -Xms200G 没有加-Xss参数，make verilog NUM_CORES=4也能正常编译，但是make emu或者simv会出现上述错误

meiqin0, 2024-06-14:
> > It's really strange... Have you tried to `make clean` or re-clone the XiangShan project?
> > 这就很怪了。您尝试过 `make clean` 或者重新 clone 香山项目吗？
> 
> 尝试过make clean && rm -rf out ，也尝试过重新 clone代码。我刚才尝试，将jvm参数改成-Xmx200G -Xms200G 没有加-Xss参数，make verilog NUM_CORES=4也能正常编译，但是make emu或者simv会出现上述错误

make emu/simv 会出现java ClassNotFoundException, 但是检查发现对应的object都存在，更改Xmx参数大小还会导致报错的objec不一样，不知道哪里出现的问题

closed 2025-09-10 by Yan-Muzi