OpenXiangShan/XiangShan #3123: ./build/emu

mlabaf2, 2024-07-01:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

hi 
I create my add.bin file from assembly file by "riscv64-linux-gnu-gcc ". now when I "./build/emu -b 0 -e 0 -i ./ready-to-run/add.bin --diff ./ready-to-run/riscv64-nemu-interpreter-so" I receive below error about start address.
[image]

### What did you do before

hi 
I create my add.bin file from assembly file by "riscv64-linux-gnu-gcc ". now when I "./build/emu -b 0 -e 0 -i ./ready-to-run/add.bin --diff ./ready-to-run/riscv64-nemu-interpreter-so" I receive below error about start address.

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

cebarobot, 2024-07-01:
How do you create your binary file? Could you please describe the details of your operation?

XiangShan emu could not accept elf file, which is the direct product of gcc. XiangShan could only accept plain binary file, which is actually the memory mirror from 0x80000000. If you want to run baremetal programs, it's neccessary to use a custom lds file to guide the linking, and it's neccessary to use objcopy to generate plain binary file.

XiangShan provides an framework for baremetal program, [nexus-am](https://github.com/OpenXiangShan/nexus-am). There is also a document that explains how to build your custom programs, [Use AM to generate custom workload](https://xiangshan-doc.readthedocs.io/zh-cn/latest/tools/gen-workload-with-am-en/).

poemonsense, 2024-07-01:
NEMU so seems wrong because it should never abort due to illegal instructions when cosim.

In https://github.com/OpenXiangShan/NEMU/blob/master/configs/riscv64-xs-ref_defconfig#L144:

```
# CONFIG_REPORT_ILLEGAL_INSTR is not set
```

mlabaf2, 2024-07-02:
> How do you create your binary file? Could you please describe the details of your operation?
>
> XiangShan emu could not accept elf file, which is the direct product of gcc. XiangShan could only accept [...]

thank you for your replay.
what is this error to run the instruction you said after "make ARCH=riscv64-xs":
make: *** No rule to make target '/Makefile.app'.  Stop.

mlabaf2, 2024-07-02:
> How do you create your binary file? Could you please describe the details of your operation?
>
> XiangShan emu could not accept elf file, which is the direct product of gcc. XiangShan could only accept [...]

I created .bin file by   
"riscv64-linux-gnu-gcc -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -Iriscv-tests/env/p/ -Iriscv-tests/isa/macros/scalar -Triscv-tests/env/p/link.ld riscv-tests/isa/rv64ui/add.S -o add.o"

cebarobot, 2024-07-02:
> thank you for your replay. what is this error to run the instruction you said after "make ARCH=riscv64-xs": make: *** No rule to make target '/Makefile.app'. Stop.

Please make sure you have set `AM_HOME` environment variable. You could set it as below:
```
# Now in nexus-am folder
export AM_HOME=$(pwd)
```
After that, you could get into `app/hello/` dir and `make ARCH=riscv64-xs`.

cebarobot, 2024-07-02:
> I created .bin file by "riscv64-linux-gnu-gcc -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -Iriscv-tests/env/p/ -Iriscv-tests/isa/macros/scalar -Triscv-tests/env/p/link.ld riscv-tests/isa/rv64ui/add.S -o add.o"

This is really close to the success. You have to use `riscv64-linux-gnu-objcopy` to generate the plain binary image:
```
riscv64-linux-gnu-objcopy -O binary add.o add.bin
```

Then you could use `add.bin` as workload image to run XiangShan.

There are a few other things you need to pay attention to:

Check the elf file using `riscv64-linux-gnu-objdump` to make sure there is no `.note.gnu.build-id` section at begin. 
```
riscv64-linux-gnu-objdump -D add.o >add.txt
```
I found that some compiler add this `.note.gnu.build-id` section. You could add `-Wl,--build-id=none` to your gcc commands to remove it.

Run your workloads on [NEMU](https://github.com/OpenXiangShan/NEMU) before XiangShan emu, so that you could find the problem of your workload more easily. You could follow [this document](https://xiangshan-doc.readthedocs.io/zh-cn/latest/tools/xsenv-en/#run-workload-using-nemu-emulator) to do that.

Modify the Pass/Fail Macro in `riscv-tests/env/p/riscv_test.h`. There is no software to handle ecall on baremetal. We have to use `nemu_trap` custom command to stop emulation.
```
#define RVTEST_PASS                                                     \
        fence;                                                          \
        li a0, 0;                                                       \
        .word 0x6b

#define RVTEST_FAIL                                                     \
        fence;                                                          \
        li a0, 1;                                                       \
        .word 0x6b
```

Good luck.

mlabaf2, 2024-07-03:
thanks for the response 
I follow the instructions , first change riscv-tests/env/p/riscv_test.h: 
[image]

after, I recieved below error, it seems above change is not correct:
[image]

cebarobot, 2024-07-03:
@mlabaf2 Sorry, your screenshot is not clear and I can't read it.

cebarobot, 2024-07-03:
@mlabaf2 
I know what's wrong. Please keep this origin macro, which is just above the original `RVTEST_FAIL` define.
```
#define TESTNUM gp
```

mlabaf2, 2024-07-03:
thanks , result is ok

closed 2024-07-03 by cebarobot