OpenXiangShan/XiangShan #3144: xsdebug

mlabaf2, 2024-07-03:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

hi
I added  --enable-log to DEBUG_ARGS in the Makefile to see report of XSDebug which are existed at the enf of DecodeUnit.scala of XiangShan, but nothing added in log file by "./build/emu -b 0 -e 0 -i ./ready-to-run/and.bin --diff ./ready-to-run/riscv64-nemu-interpreter-so -b 1 -e 11000 2>&1 | tee > MY_log_add_capstone.log" 

### What did you do before

 "./build/emu -b 0 -e 0 -i ./ready-to-run/and.bin --diff ./ready-to-run/riscv64-nemu-interpreter-so -b 1 -e 11000 2>&1 | tee > MY_log_add_capstone.log"

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

mlabaf2, 2024-07-04:
before  ./build/emu, I have an assembly code which is converted to and.bin

and what is the difference between xsdebug and print instruction in log result?
[image]

can I see both of them in log file by ./build/emu instruction?

cebarobot, 2024-07-04:
`XSDebug` is used for printing when **running emu**. It's based on `printf` of chisel. You need to re-compile the emu (`make emu ...` again) to use that.

`print` is used for printing when **compiling chisel**. It's a scala built-in.

mlabaf2, 2024-07-04:
thank you for the response. 
XSDebug is ok now. thanks. 
how can I see result of compiling chisel to check print?

cebarobot, 2024-07-04:
It will print out when running `make emu ...`. Please check the output in the terminal.

mlabaf2, 2024-07-04:
but during make emu...., we still do not specify any input file for bare metal to execute and as a result print some note from code? for example in chisel I specify print add instruction when assemble code is add.....

cebarobot, 2024-07-04:
The scala built-in `print` could only print the infomation about the **static structure of circuits**, when compilating chisel to verilog. It's NOT used for dynamic state of the circuits during emulating.

If you want to print something during emulating to show some dynamic state of the processor circuits, please consider to use `XSDebug` (provided by XiangShan, based on `printf`), or directly use `printf` (provide by chisel).

mlabaf2, 2024-07-04:
thank you for response. what is the necessary library for each of them and the format of them and location to insert in code for each of them ....print and printf?

cebarobot, 2024-07-04:
Please refer to existing code and [chisel Document](https://www.chisel-lang.org/docs/explanations/printing).

mlabaf2, 2024-07-05:
thanks for your help

closed 2024-07-05 by cebarobot