OpenXiangShan/XiangShan #3153: invalid opcode

mlabaf2, 2024-07-06:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

HI
i added some new opcode to XiangShan. I modify Instruction.scala, ALU.scala, parameters.scala, DecodeUnit.scala. when I run my workspace , I recieved this message in report of .log file . I cannot find from which .scala code this message is produced "invalid opcode...."
[image]

mlabaf2, 2024-07-06:
Another question is which .scala file call class Alu in Alu.scala file? I cannot find it to trace the mentioned error message...

cebarobot, 2024-07-06:
This invalid opcode message is come from [NEMU](https://github.com/OpenXiangShan/NEMU), the golden reference model of XiangShan. You may modify the NEMU to support your new instruction, or just add `--no-diff` argument to run emu in no diff mode.

cebarobot, 2024-07-06:
> Another question is which .scala file call class Alu in Alu.scala file? I cannot find it to trace the mentioned error message...

I think it's here. 

https://github.com/OpenXiangShan/XiangShan/blob/c2ce03f70f78cf3e4a3588a9396eaf9b67952cd9/src/main/scala/xiangshan/backend/fu/FuConfig.scala#L297-L307

And this `fuGen` is called here.

https://github.com/OpenXiangShan/XiangShan/blob/7ffbf5fd74959ca8274ee07f523682ef2495c1d6/src/main/scala/xiangshan/backend/exu/ExeUnit.scala#L62-L66

mlabaf2, 2024-07-29:
thank you . the problem is solved. now I add some IO ports to 
FuOutput and FunctionUnitInput.

 [image]

although I dontcare them in  FunctionUnit, I recived below error. I dont undrestand why and must manipulate which part?
[image]

[image]

[image]
[image]

mlabaf2, 2024-07-30:
thanks solved

closed 2025-09-05 by Yan-Muzi