OpenXiangShan/XiangShan #3154: error massage

mlabaf2, 2024-07-07:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

Hi
my XiangShan has build folder with emu folder in it. 
previously it was working but now, when I want to make emu... I encounter this error:
[image]

### Expected behavior

I use  export LC_ALL=C for local setting

### To Reproduce

I use  export LC_ALL=C for local setting

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id:

Tang-Haojin, 2024-07-08:
It seems that there are only some warnings but no error. Please check if there is already an emu in ./build path.

mlabaf2, 2024-07-08:
Again I make verilog and problem fixed. Is it neccesary each time we change code, both make verilog and then make emu?

cebarobot, 2024-07-09:
> Again I make verilog and problem fixed. Is it neccesary each time we change code, both make verilog and then make emu?

No, I think `make emu` is enough... `make` should arrange the dependency automatically.

The before log is not an error. `make` thinks that there is no file modified and emu is up-to-date, so it does not need to do anything. Please make sure that you did modify some source files.

closed 2025-09-05 by Yan-Muzi