OpenXiangShan/XiangShan #3155: What is the stable version of Nanhu-V2 verified on FPGA?

xuyb1999, 2024-07-08:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

您好，请问有在FPGA上验证通过的南湖V2的稳定版本吗？我看到release里面有两个v2.0和v2.1的tag，它们哪一个是通过FPGA验证，哪一个是流片的南湖V2版本呢？

Hello, may I ask if there is a stable version of Nanhu V2 that has been verified on FPGA? I noticed there are two tags, v2.0 and v2.1, in the release. Which one has been verified on FPGA, and which one is the taped-out version of Nanhu V2?

Tang-Haojin, 2024-07-08:
The southlake branch or tag v2.1 is Nanhu V2. To place it on FPGA vu19p, you may referred to [this patch](https://github.com/OpenXiangShan/XiangShan/commit/4951141ce593f31024884bb9e150e5519f8cf913).

Tang-Haojin, 2024-07-09:
You may try not to modify `class DefaultConfig` or delete `FPGAConfig`, and use `make verilog CONFIG=FPGAConfig`. It seems that this is already an FPGA-adapted config and you can use it.

xuyb1999, 2024-07-10:
I have tried to use the original `FPGAConfig`, and it passes `make verilog CONFIG=FPGAConfig`. But when I use [OpenXiangShan/env-scripts](https://github.com/OpenXiangShan/env-scripts) (commit ID 506e9e5) to generate the bitstream for Nanhu-V2.1, the procedure fails at synthesis. 

Here is the output errors of the synthesis runme.log:

~~~text
WARNING: [Synth 8-7071] port 'SOC_M_AXI_arregion' of module 'jtag_ddr_subsys_wrapper' is unconnected for instance 'U_JTAG_DDR_SUBSYS' [/nfs/home/user/Workspace/env-scripts/xs_nanhu_fpga/src/rtl/xs_core_def_xdma.v:1366]
WARNING: [Synth 8-7071] port 'SOC_M_AXI_awregion' of module 'jtag_ddr_subsys_wrapper' is unconnected for instance 'U_JTAG_DDR_SUBSYS' [/nfs/home/user/Workspace/env-scripts/xs_nanhu_fpga/src/rtl/xs_core_def_xdma.v:1366]
WARNING: [Synth 8-7023] instance 'U_JTAG_DDR_SUBSYS' of module 'jtag_ddr_subsys_wrapper' has 61 connections declared, but only 59 given [/nfs/home/user/Workspace/env-scripts/xs_nanhu_fpga/src/rtl/xs_core_def_xdma.v:1366]
INFO: [Synth 8-6157] synthesizing module 'XSTop_wrapper' [/nfs/home/user/Workspace/env-scripts/xs_nanhu_fpga/src/rtl/XSTop_wrapper.v:1]
ERROR: [Synth 8-439] module 'XSTop' not found [/nfs/home/user/Workspace/env-scripts/xs_nanhu_fpga/src/rtl/XSTop_wrapper.v:161]
ERROR: [Synth 8-6156] failed synthesizing module 'XSTop_wrapper' [/nfs/home/user/Workspace/env-scripts/xs_nanhu_fpga/src/rtl/XSTop_wrapper.v:1]
ERROR: [Synth 8-6156] failed synthesizing module 'xs_core_def' [/nfs/home/user/Workspace/env-scripts/xs_nanhu_fpga/src/rtl/xs_core_def_xdma.v:3]
ERROR: [Synth 8-6156] failed synthesizing module 'xs_fpga_top_debug' [/nfs/home/user/Workspace/env-scripts/xs_nanhu_fpga/src/rtl/xs_fpga_top_debug.v:5]
---------------------------------------------------------------------------------
Finished RTL Elaboration : Time (s): cpu = 00:00:14 ; elapsed = 00:00:32 . Memory (MB): peak = 4501.078 ; gain = 1486.586 ; free physical = 92541 ; free virtual = 139143
---------------------------------------------------------------------------------
RTL Elaboration failed
INFO: [Common 17-83] Releasing license: Synthesis
85 Infos, 28 Warnings, 0 Critical Warnings and 5 Errors encountered.
synth_design failed
ERROR: [Common 17-69] Command failed: Synthesis failed - please see the console or run log file for details
INFO: [Common 17-206] Exiting Vivado at Wed Jul 10 11:36:49 2024...
~~~

My procedures to use OpenXiangShan/env-scripts are as follows:

~~~bash
cd env-scripts/xs_nanhu_fpga
make update_core_flist CORE_DIR=`realpath ../../Nanhu-v2.1/build`
make nanhu CORE_DIR=`realpath ../../Nanhu-v2.1/build`
make bitsream CORE_DIR=`realpath ../../Nanhu-v2.1/build`
~~~

Many thanks to your kindly help!

xuyb1999, 2024-07-10:
Moreover, I have tried to modify the `FPGAConfig` to set `AlwaysBasicDiff` and `sramClkDivBy2` to false, but the same synthesis errors still happen.

~~~diff
diff --git a/src/main/scala/top/Configs.scala b/src/main/scala/top/Configs.scala
index c10650129..89763c6c1 100644
--- a/src/main/scala/top/Configs.scala
+++ b/src/main/scala/top/Configs.scala
@@ -318,8 +318,17 @@ class DefaultConfig(n: Int = 1) extends Config(
 )

 class FPGAConfig(n: Int = 1) extends Config(
-  new WithNKBL3(2 * 1024, inclusive = false, banks = 4, ways = 8)
+  (new WithNKBL3(2 * 1024, inclusive = false, banks = 4, ways = 8)
     ++ new WithNKBL2(512, inclusive = false, banks = 2, alwaysReleaseData = true)
     ++ new WithNKBL1D(128)
-    ++ new BaseConfig(n)
+    ++ new BaseConfig(n)).alter((site, here, up) => {
+    case DebugOptionsKey => up(DebugOptionsKey).copy(
+      AlwaysBasicDiff = false
+    )
+    case SoCParamsKey => up(SoCParamsKey).copy(
+      L3CacheParamsOpt = Some(up(SoCParamsKey).L3CacheParamsOpt.get.copy(
+        sramClkDivBy2 = false,
+      ))
+    )
+  })
 )
~~~

Tang-Haojin, 2024-07-10:
Seems that it did not found XSTop module. You may rename generated XSTop.v to XSTop.sv and add it to the FPGA project manually.

xuyb1999, 2024-07-18:
Sorry for my late reply. I follow your instructions to rename XSTop.v to XSTop.sv, execute the procedures of env-scripts, and find XSTop.sv is already added to the FPGA project. 

[image]

Then I try to update the out-of-date hierarchy in the vivado project by "hierarchy->reload->Hierarchy Update -> Automatical Update and Compile Order", and specify a new top module. However, no matter I select the `xs_fpga_top_debug` or `bosc_XSTop`, the same synthesis error will still occur.

Could this be due to the env-scripts version being incompatible with the current Nanhu-v2.1?

Tang-Haojin, 2024-07-18:
Seems that the module in the Hierarchy is `bosc_XSTop`. Did you add prefix argument? If so, you may also change the module name in `XSTop_wrapper`.

xuyb1999, 2024-07-31:
By change the module name in OpenXiangShan/env-scripts, the bitstream of Nanhu-V2.1 is successfully generated.

Thank you for your kind help!

~~~diff
diff --git a/xs_nanhu_fpga/src/rtl/XSTop_wrapper.v b/xs_nanhu_fpga/src/rtl/XSTop_wrapper.v
index d300425..7fc6976 100755
--- a/xs_nanhu_fpga/src/rtl/XSTop_wrapper.v
+++ b/xs_nanhu_fpga/src/rtl/XSTop_wrapper.v
@@ -158,7 +158,7 @@ module XSTop_wrapper(
 assign cpu_to_soc = 32'h0;

-XSTop  u_XSTop(
+bosc_XSTop  u_XSTop(
   .memory_0_awready                (mem_core_awready )                        ,
   .memory_0_awvalid                (mem_core_awvalid )                        ,
   .memory_0_awid                   (mem_core_awid    )                          ,
~~~

closed 2024-07-31 by xuyb1999