OpenXiangShan/XiangShan #3195: 在使用docker检查编译app生成data.txt，烧录后找不到编译的指令

LOCKEDGATE, 2024-07-12:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

按照这个文档https://xiangshan-doc.readthedocs.io/zh-cn/latest/integration/docker/
进行操作，我先用docker交叉编译了一个c文件，编译器就是riscv64-unknown-linux-gnu-gcc，然后将这个编译后的二进制文件按照文档压缩到rootfs.cpio中，其他按文档操作。生成data.txt后，系统烧录成功了，但是在/usr/bin下找不到我编译的文件

### What did you do before

如上所示

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

Tang-Haojin, 2024-07-14:
请进一步检查你生成的cpio文件中是否含有你编译的文件，如果没有，说明可能你编译的文件没有正确地放入cpio包中。

Please further check that the CPIO file you generated contains the files you compiled. If not, it means that the files you compiled may not have been properly placed in the CPIO package.

LOCKEDGATE, 2024-07-14:
我手动创建了一个C文件，内容就是打印五次hello，然后我在docker中用riscv-gcc交叉编译了它，命名为hello5，把它放入cpio的/usr/bin下，我还解压了cpio，确保目录下有该文件，但是在烧录后，在系统下的/usr/bin下找不到该文件，原因有可能出现在我创建的hello5文件上吗，还是说，只要我往cpio中放入任何文件，烧录后该文件都会存在。
I manually created a C file that prints hello five times, then I cross-compiled it in docker with riscv-gcc, named it hello5, and put it in /usr/bin of cpio. I also unzipped cpio to make sure the file exists in the directory, but after burning, I can't find the file in /usr/bin under the system. Could the reason be the hello5 file I created, or does it mean that as long as I put any file in cpio, the file will exist after burning?

cebarobot, 2024-07-14:
如果您自定义了 cpio 的 initramfs，您可能需要在 kernel 的配置项 `CONFIG_INITRAMFS_SOURCE` 中指定您的 cpio 文件的地址。
If you have customized the initramfs with cpio, you may need to specify the location of your cpio file in the kernel configuration option `CONFIG_INITRAMFS_SOURCE`.

在 `riscv-linux` 目录下执行：
Execute the following in the riscv-linux directory:
```
make ARCH=riscv menuconfig
```

选择 `General setup` 菜单，选择 `Initramfs source file` 配置项，将其改为您的 cpio 文件的路径。
Select the General setup menu, choose the Initramfs source file option, and change it to the path of your cpio file.

然后再次编译 kernel：
Then recompile the kernel:
```
cd ../
make sw -j200  #完成后buid目录下有linux.bin
```

LOCKEDGATE, 2024-07-15:
感谢，我尝试了您的方法，在Initramfs source file中指定了我的cpio，但是还是不行。我其实就是在/usr/bin下添加了一个交叉编译后的C文件hello5，它的内容就是打印五次hello，但是我在烧录data.txt后，启动系统发现文件夹下没有我的hello5。下图就是我创建的hello5文件，并放在了/usr/bin中。
Thanks, I tried your method and specified my cpio in the Initramfs source file, but it still didn't work. I actually added a cross-compiled C file hello5 under /usr/bin, and its content was to print hello five times, but after burning data.txt, I started the system and found that my hello5 was not in the folder. The picture below shows the hello5 file I created and placed in /usr/bin.
[image]
我为了排除是不是我的文件错误了，我把cpio中自带的指令yes，如下图所示，它的功能是无线打印y这个字符，我把这个指令yes删除，然后复制我的hello5并更名为yes，在烧录后，我执行yes，发现打印的还是y，这好像证明，我无论怎么修改cpio文件，都无法改变它原来的内容。
In order to rule out the possibility that my file was wrong, I deleted the command yes in cpio, as shown in the figure below. Its function is to wirelessly print the character y. I copied my hello5 and renamed it yes. After burning, I executed yes and found that it was still printed as y. This seems to prove that no matter how I modify the cpio file, I cannot change its original content.
[image]

LOCKEDGATE, 2024-07-15:
非常感谢您的帮助，流程已经调通了！
Thank you very much for your help, the process has been adjusted！

closed 2024-07-16 by Tang-Haojin