OpenXiangShan/XiangShan #3248: How to use other versions of Linux kernel for Xiangshan projects？

LOCKEDGATE, 2024-07-19:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

我按照[https://xiangshan-doc.readthedocs.io/zh-cn/latest/integration/docker/](url)所述，跑通了流程，十分感谢。但是该项目使用的linux内核版本为linux-5.10.167，我查看[https://github.com/OpenXiangShan/riscv-linux](url)下的其他分支，其中还包括了一些其他版本的linux内核。假如我现在希望将其他版本的原生的linux内核[https://github.com/riscvarchive/riscv-linux](url)，比如linux-6.1.39，构建为ns-bbl/riscv-linux，我该怎么将它转换为xiangshan能用的内核，或者说与之有关的文档被我忽略掉的。十分感谢。

I followed the instructions in [https://xiangshan-doc.readthedocs.io/zh-cn/latest/integration/docker/](url) and ran through the process. Thank you very much. However, the linux kernel version used in this project is linux-5.10.167. I checked other branches under [https://github.com/OpenXiangShan/riscv-linux](url), which also includes some other versions of the linux kernel. If I now want to build other versions of the native linux kernel [https://github.com/riscvarchive/riscv-linux](url), such as linux-6.1.39, into ns-bbl/riscv-linux, how can I convert it to a kernel that can be used by Xiangshan, or the relevant documents have been ignored by me.Thank you very much.

### What did you do before

我已经按照文档所述的内容[https://xiangshan-doc.readthedocs.io/zh-cn/latest/integration/docker/](url)跑通了，跑通版本为linux-5.10.167.

I have followed the documentation [https://xiangshan-doc.readthedocs.io/zh-cn/latest/integration/docker/](url) and the running version is linux-5.10.167.

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

cebarobot, 2024-07-19:
香山可以直接使用6.0以后版本的原版内核，不需要额外修改内核代码。您可以尝试直接用指定版本的内核源码代替原先的riscv-linux文件夹。

另外，目前我们已经逐步从riscv-pk切换到openSBI。您可参考opensbi的文档，将内核作为payload编入opensbi中，然后直接用作香山的workload。这部分文档我们还在建设中，您也可参考目前的草稿：https://github.com/OpenXiangShan/XiangShan-doc/blob/79e02cbeb6bf3bbdc920225816fd127e8549b794/docs/tools/opensbi-kernel-for-xs.md

LOCKEDGATE, 2024-07-19:
十分感谢，另外，如果我想裁剪内核，比如网口是我的FPGA所不需要的。我需要修改哪些配置文件？

Thank you very much. In addition, if I want to fix the kernel, for example, the network port is not needed by my FPGA, which configuration files do I need to modify?

cebarobot, 2024-07-19:
您可以使用 menuconfig 对内核进行裁剪。在内核源码目录下运行：`make ARCH=riscv menuconfig`。具体请您参考互联网上有关内核配置的资料。

LOCKEDGATE, 2024-07-19:
十分感谢您的回复，对我受益匪浅。
Thank you very much for your reply, it is very helpful to me

closed 2024-07-19 by LOCKEDGATE