OpenXiangShan/XiangShan #3271: How can I check the CPU frequency of the two cores of Xiangshan?

LOCKEDGATE, 2024-07-23:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

您好，我在尝试利用dhrystone程序检验xiangshan的cpu性能，其中一项参数需要cpu的主频。我使用了xiangshan两core的cpu，我尝试去/proc/cpuinfo下查看相关信息，但是该文件中没有详细描述cpu的参数。那么我该怎么查看其主频？谢谢。
Hello, I am trying to use the dhrystone program to check the CPU performance of Xiangshan. One of the parameters requires the CPU's main frequency. I use a two-core CPU of Xiangshan. I try to check the relevant information under /proc/cpuinfo, but the file does not describe the CPU parameters in detail. So how can I check its main frequency? Thank you.

### What did you do before

尝试去/proc/cpuinfo下查看相关信息

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

LOCKEDGATE, 2024-07-24:
尝试利用lscpu查看，但是打印的信息和查看/proc/cpuinfo的内容相差不大

cebarobot, 2024-07-24:
在仿真环境下，香山处理器的主频是一个外部给的定值，目前昆明湖架构通常指定为 3 GHz。这个数据应该通过设备树传入 linux，但应该不能通过 `/proc/cpuinfo` 查看。

In emulation environment, the frequence of XiangShan is an constants, which is always set to 3 GHz in Kunminghu. This parameter is passed to linux through device tree, and may not be shown in `/proc/cpuinfo`.

LOCKEDGATE, 2024-07-24:
我查看了https://github.com/OpenXiangShan/riscv-pk/tree/noop/dts下的设备树文件夹，请问如果要加入CPU主频，应该往哪个文件里加入？
I checked the device tree folder under https://github.com/OpenXiangShan/riscv-pk/tree/noop/dts. If I want to add the CPU main frequency, which file should I add it to?

LOCKEDGATE, 2024-07-24:
我查看了[https://github.com/OpenXiangShan/riscv-pk/tree/noop/dts下的设备树文件夹，请问如果要加入CPU主频，应该往哪个文件里加入？
I checked the device tree folder under https://github.com/OpenXiangShan/riscv-pk/tree/noop/dts. If I want to add the CPU main frequency, which file should I add it to?

cebarobot, 2024-07-24:
应当加在 `system.dts` 中 `cpu@0` 的 `clock-frequency` 配置项，具体可参考：https://www.kernel.org/doc/Documentation/devicetree/booting-without-of.txt

另外，我不确定 dhrystone 是如何计时的。如果它是通过 `mcycle` 或者 `cycle` 来计时，那么参考 CPU 主频是有用的。如果它是通过 `time.h` 或者 `clock()` 来计时的，它大概率依赖的是 `mtime` 或 `time`，使用它们需要设置 dts 中的 `timebase-frequency` 项。

It should be described in `clock-frequency`, which is in `cpu@0` of `system.dts`. Please refer to https://www.kernel.org/doc/Documentation/devicetree/booting-without-of.txt

By the way, I'm not sure about how dhrystone get the time. If it uses `mcycle` or `cycle`, it's ok to refer to CPU base frequence. If it uses `time.h` or `clock()`, which is relay on `mtime` and `time`, then you should setup the `timebase-frequency` in dts.

LOCKEDGATE, 2024-07-25:
好的感谢，我尝试一下
Thanks，I‘ll try it.

closed 2024-07-25 by LOCKEDGATE