OpenXiangShan/XiangShan #3289: Run spec2006 using opensbi-kernel-for-xs tutorial

LOCKEDGATE, 2024-07-25:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

您好，我在通过教程 https://github.com/OpenXiangShan/XiangShan-doc/blob/79e02cbeb6bf3bbdc920225816fd127e8549b794/docs/tools/opensbi-kernel-for-xs.md 到这一步骤：将 initramfs 的 source 改为 $RISCV_ROOTFS_HOME/rootfsimg/initramfs-spec.txt，我在make时报错如下
[image]
我查看了initramfs-spec.txt里的内容，发现如下的路径中，对应的文件不存在。请问我是需要手动下载spec2006并设置${SPEC_BUILD_PATH}与${CPU2006_RUN_DIR}吗？
[image]

Hello, I am following the tutorial https://github.com/OpenXiangShan/XiangShan-doc/blob/79e02cbeb6bf3bbdc920225816fd127e8549b794/docs/tools/opensbi-kernel-for-xs.md to this step: Change the source of initramfs to $RISCV_ROOTFS_HOME/rootfsimg/initramfs-spec.txt, and I get the following error in make.
I checked the content of initramfs-spec.txt and found that the corresponding file does not exist in the following path. Do I need to manually download spec2006 and set ${SPEC_BUILD_PATH} and ${CPU2006_RUN_DIR}?

### What did you do before

如上

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

cebarobot, 2024-07-25:
是的，您需要自行准备 SPEC CPU 2006 的相关文件。`initramfs-*.txt` 不是开箱即用的，需要您自己做调整。您可另参考我们的视频教程：https://www.bilibili.com/video/BV1Wr421h7XN/

Yes, you need to prepare SPEC CPU 2006 files by yourself. `initramfs-*.txt` is not out-of-the-box. You have to modify it for your environment. Also, you could refer to our video tutorial (in Chinese): https://www.bilibili.com/video/BV1Wr421h7XN/

LOCKEDGATE, 2024-07-25:
好的。我尝试一下。
另外，假如我交叉编译了spec 2006，能把它作为自定义app按照https://xiangshan-doc.readthedocs.io/zh-cn/latest/integration/docker/里的指令烧录到cpio文件里，然后再上板验证吗？
OK. I'll try it.
In addition, if I cross-compile spec 2006, can I use it as a custom app and burn it into the cpio file according to the instructions in https://xiangshan-doc.readthedocs.io/zh-cn/latest/integration/docker/, and then verify it on the board?

LOCKEDGATE, 2024-08-05:
您好，我尝试交叉编译好了spec2006，然后我按照https://xiangshan-doc.readthedocs.io/zh-cn/latest/integration/docker/把整个spec2006它放进了rootfs.cpio，生成的data.txt大小为1.4G，我也看到了https://xiangshan-doc.readthedocs.io/zh-cn/latest/tools/linux-kernel-for-xs/ 描述的修改设备树来添加地址空间，我先用了个没有添加spec2006的cpio进行测试，目的是验证我修改设备树后地址空间有没有生效，然后运行成功。
接着我使用包含spec2006的1.4G的cpio包，烧录后运行卡住
[yunxing_0802.txt](https://github.com/user-attachments/files/16490221/yunxing_0802.txt)
我想明确利用cpio的方法移植spec2006是可行的吗
Hello, I tried to cross-compile spec2006, and then I put the entire spec2006 into rootfs.cpio, and the generated data.txt size was 1.4G. I also saw https://xiangshan-doc.readthedocs.io/zh-cn/latest/tools/linux-kernel-for-xs/ describing how to modify the device tree to add address space. I first used a cpio without adding spec2006 to test, the purpose was to verify whether the address space took effect after I modified the device tree, and then it ran successfully.
Then I used the 1.4G cpio package containing spec2006, and it got stuck after burning
I want to know whether it is feasible to transplant spec2006 using the cpio method

cebarobot, 2024-08-12:
这样做是可行的。看起来您的kernel完全没有打印启动日志，这可能是多方面原因造成的，还需要仔细检查各个细节。

This is feasible. It seems that your kernel didn't print out any boot logs, which may be caused by multiple reasons. More check are needed.

closed 2024-08-14 by LOCKEDGATE