OpenXiangShan/XiangShan #3346: How can I make a 4core image?

LOCKEDGATE, 2024-08-06:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

你好，我按照 https://xiangshan-doc.readthedocs.io/zh-cn/latest/integration/docker/ 想制作一个4core的image，我把 
 https://github.com/xyyy1420/workload_build_env/blob/master/dts/fpga-fourcore-system.dts 这个文件直接拷过去放进riscv-pk下的dts中，替换为原来的system.dts，制作完镜像后，上板烧录时乱码。
接着，我利用 https://github.com/OpenXiangShan/XiangShan-doc/blob/79e02cbeb6bf3bbdc920225816fd127e8549b794/docs/tools/opensbi-kernel-for-xs.md 描述的，制作了一个4core的fw_payload.bin，将它转换为data.txt，并上板测试，发现无任何打印信息。
所以我想请教一下，如果按照docker的教程走，能否生成一个4core的image？或者说在https://github.com/OpenXiangShan/XiangShan-doc/blob/79e02cbeb6bf3bbdc920225816fd127e8549b794/docs/tools/opensbi-kernel-for-xs.md 里，我操作不当？

Hello, I want to make a 4-core image according to https://xiangshan-doc.readthedocs.io/zh-cn/latest/integration/docker/. I copied the file https://github.com/xyyy1420/workload_build_env/blob/master/dts/fpga-fourcore-system.dts directly into the dts under riscv-pk and replaced it with the original system.dts. After making the image, garbled characters appeared when burning on the board. Then, I made a 4-core fw_payload.bin using the description in https://github.com/OpenXiangShan/XiangShan-doc/blob/79e02cbeb6bf3bbdc920225816fd127e8549b794/docs/tools/opensbi-kernel-for-xs.md, converted it to data.txt, and tested it on the board, but found that there was no print information.
So I would like to ask, if I follow the docker tutorial, can I generate a 4-core image? Or in https://github.com/OpenXiangShan/XiangShan-doc/blob/79e02cbeb6bf3bbdc920225816fd127e8549b794/docs/tools/opensbi-kernel-for-xs.md, did I do something wrong?

### What did you do before

如上

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

LOCKEDGATE, 2024-08-12:
在上述的基础上，我尝试将南湖v3的pk放进  https://xiangshan-doc.readthedocs.io/zh-cn/latest/integration/docker/ 这里，替换掉原来的pk，以期望生成一个4core的镜像，我在上板时打印如下：
[yunxing_4core.txt](https://github.com/user-attachments/files/16577075/yunxing_4core.txt)
请问这个问题该如何解决

cebarobot, 2024-08-12:
我不太明白您所说的“上板烧录时乱码”是指什么。

至于 `opensbi-kernel-for-xs.md` 中提到的四核设备树，其面向的是软件仿真环境，UART 等设备配置与集成指南提供的 FPGA 平台（以及您使用的 FPGA 平台）有着明显的差异。您需要手动调整相关设备树配置来在 FPGA 上使用。

I'm not sure what you mean by "garbled characters appeared when burning on the board". 

The four core device tree mentioned in `opensbi-kernel-for-xs.md` is just used for software emulation environment, which has huge different in devices such as UART. Please modify dts manually.

LOCKEDGATE, 2024-08-12:
感谢您的回复，在上述的基础上，我尝试将南湖v3的pk放进 https://xiangshan-doc.readthedocs.io/zh-cn/latest/integration/docker/ 这里，替换掉原来的pk，以期望生成一个4core的镜像，这似乎比我直接用之前的面向仿真的4核设备树更加贴切，我在上板时打印如下：
[yunxing_4core.txt](https://github.com/user-attachments/files/16577075/yunxing_4core.txt)
请问这个问题该如何解决？我直接整体直接替换掉pk的做法，是行得通的吗

Thank you for your reply. Based on the above, I tried to put the pk of Nanhu v3 into https://xiangshan-doc.readthedocs.io/zh-cn/latest/integration/docker/ here to replace the original pk, hoping to generate a 4core image. This seems to be more appropriate than using the previous simulation-oriented 4-core device tree directly. When I put it on the board, it prints as follows:
[yunxing_4core.txt](https://github.com/user-attachments/files/16577075/yunxing_4core.txt)
How can I solve this problem? Is it feasible for me to directly replace the pk as a whole?

cebarobot, 2024-08-12:
> 在上述的基础上，我尝试将南湖v3的pk放进 https://xiangshan-doc.readthedocs.io/zh-cn/latest/integration/docker/ 这里，替换掉原来的pk，以期望生成一个4core的镜像，我在上板时打印如下： [yunxing_4core.txt](https://github.com/user-attachments/files/16577075/yunxing_4core.txt) 请问这个问题该如何解决

看起来仍然是设备树相关的问题。设备树指定了某个地址存在某个设备，但香山的 PMA 认为这里没有这个设备，于是报出访问错误（Access fault）。

请注意，设备树需要根据您使用的平台进行调整和修改，不是开箱即用的。直接替换 pk 的做法大概行不通。

It still seems to be a device tree related problem. The dts says that there is a device at some address, but PMA of XiangShan thinks that there is no such device. So access fault occurs.

FYA, dts should be modified according to your platform. dts is always not out-of-the-box. I don't think replacing pk directly is a good way.

LOCKEDGATE, 2024-08-12:
那香山有没有现有的教程能够给出如何制作一个4core的image的相关文档？十分感激
Is there any existing tutorial in Xiangshan that can provide relevant documents on how to make a 4core image? Thank you very much

cebarobot, 2024-08-12:
> 那香山有没有现有的教程能够给出如何制作一个4core的image的相关文档？十分感激 Is there any existing tutorial in Xiangshan that can provide relevant documents on how to make a 4core image? Thank you very much

我们目前没有文档介绍如何制作用于 FPGA 平台的4核的工作负载的文档。
We don't have tutorial for how to make a workload image for 4-core FPGA platform.

LOCKEDGATE, 2024-08-12:
好的感谢，我尝试修改一下设备树
Thanks a lot，I'll try to modify the device tree.

LOCKEDGATE, 2024-08-16:
您好，我尝试修改了nanhuv3下的dts/system.dts，并找到了疑似报错的xdma设备，我把xdma_0和xdma_1的定义都删除了，上板测的时候发现报的错误还是一样的，我想知道是我改错设备树了吗

cebarobot, 2024-08-16:
缺少信息，不太好确定问题，但大概率还是设备树相关问题。

Lack of information makes it difficult to determine the problem. It is highly likely to be related to the device tree.

closed 2024-08-19 by LOCKEDGATE