OpenXiangShan/XiangShan #3365: 请教一下关于chi版本仿真的一个问题

spritelzj, 2024-08-09:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

请教一下关于chi版本仿真的一个问题：
1、首先我在/xiangshan/路径下执行了make verilog CONFIG=XSNoCTopConfig MFC=1生成了CHI协议的rtl，我在XSTop.sv中可以看到CHI接口，如图1所示；

2、然后我执行make eumu -j32 EMU_TRACE=fst，我发现XSTop.sv的顶层信号变了，如图2所示。这一步是不是需要加其他参数？

[image]
        图1
[image]
        图2 

### What did you do before

按照https://xiangshan-doc.readthedocs.io/zh-cn/latest/tools/xsenv/所描述步骤进行了环境的搭建

### Environment

- XiangShan branch: master
- XiangShan commit id: commit a061b23e782d765edb3bd0f84158b8066175334d (HEAD)
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

cebarobot, 2024-08-12:
`make verilog` 和 `make emu` 两个步骤的目标不同，前者生成用于综合的 Verilog 代码，后者生成用于仿真的可执行文件。两者没有流程上的关联，不共用同一套 Verilog。

如果您想针对 `XSNoCTopConfig` 这个 config 进行软件仿真，请直接在 `make emu` 时加入 `CONFIG=XSNoCTopConfig` 参数。

`make verilog` and `make emu` has different target. `make verilog` is used for generating Synthesizable Verilog codes. `make emu` is used for generating software emulation excetuable file. There is no procedural connection between them and they uses different verilog codes.

If you want to emualte `XSNoCTopConfig`, please add `CONFIG=XSNoCTopConfig` in command `make emu`.

spritelzj, 2024-08-12:
我使用make emu CONFIG=XSNoCTopConfig MFC=1生成的仿真文件进行仿真后，波形中的代码与make verilog CONFIG=XSNoCTopConfig MFC=1生成的代码(build/rtl/路径下的代码)不一致？请问这是合理的吗？

After emulation using the emulation file generated by make emu CONFIG=XSNoCTopConfig MFC=1, The code in the waveform is not consistent with the code generated by make verilog CONFIG=XSNoCTopConfig MFC=1 (code under the build/rtl/ path)? Is this reasonable?

cebarobot, 2024-08-12:
> 我使用make emu CONFIG=XSNoCTopConfig MFC=1生成的仿真文件进行仿真后，波形中的代码与make verilog CONFIG=XSNoCTopConfig MFC=1生成的代码(build/rtl/路径下的代码)不一致？请问这是合理的吗？
>
> After emulation using the emulation file generated by make [...]

`make verilog` 生成**用于综合的 Verilog 代码**。`make emu` 通过 `make sim-verilog`生成**用于仿真的 Verilog 代码**，而后再通过 verilator 生成可执行文件。两者最终目标不同，**Verilog 也是不同的**。

`make verilog` generates **Verilog codes for synthesis**. `make emu` generates **Verilog codes for emulation** through `make sim-verilog`, then generates executable file by verilator. They have different purpose and **use different Verilog**.

closed 2025-09-05 by Yan-Muzi