OpenXiangShan/XiangShan #3368: Is there any tutorial on how to run XiangShan RISC-V Processor on S2C's vu19p?

yorange1, 2024-08-12:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

I saw in this article that you ran a RISC-V processor in an S2C fpga prototype (https://riscv.org/blog/2023/11/s2cs-fpga-prototyping-accelerates-iterations-of-xiangshan-risc-v-processor/)，Are there any tutorials or documentation on this?

cebarobot, 2024-08-12:
We do have an tutorial here: [FPGA Prototype - XiangShan Doc](https://xiangshan-doc.readthedocs.io/zh-cn/latest/integration/fpga/). It's all in Chinese. Some infomation or code in that tutorial may be out of date.

yorange1, 2024-08-12:
Thanks, I'll give it a try.

cebarobot, 2025-07-24:
This issue has been inactive for a while. If you have any further questions, feel free to reopen it or open a new one.