OpenXiangShan/XiangShan #3377: Is there any official OpenSBI version for booting linux on S2C's Prodigy™ S7-19PD Logic System？

stwjt, 2024-08-14:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

您好，我想了解是否有一个香山官方支持的opensbi版本能够直接支持 S2C的Prodigy™ S7-19PD Logic System 的 启动。
Hello, I would like to know if there is an official Xiangshan-supported version of OpenSBI that can directly support the booting of the S2C Prodigy™ S7-19PD Logic System.

### What did you do before

我在 GitHub 上只找到了依赖于 riscv-pk 实现的项目，并且我看到issue#3248提到“另外，目前我们已经逐步从riscv-pk切换到openSBl”。我先运行了ns-bbl的riscv-pk，bootloader能成功启动，但是linux内核没有显示，关于这个我也想问原因。之后我在opensbi上根据riscv-pk的uart_snps改了下，新增了uart_snps的一些函数，修改了libs/util/serial，uart串口能打印字符，在NEMU上能够正常运行，但在S2C的vu19p上跑似乎会在进入_scratch_init或者之后就卡死。目前不清楚原因。
On GitHub, I only found projects dependent on riscv-pk. Issue #3248 mentions 'Furthermore, we have been gradually transitioning from riscv-pk to openSBI.' Initially, I ran ns-bbl with riscv-pk and the bootloader boots successfully, but the Linux kernel does not display any output. I also want to ask about the possible reasons for this. Then, I modified openSBI based on riscv-pk's uart_snps, adding some functions for uart_snps and modifying libs/util/serial. The UART serial port can print characters, and it runs correctly in NEMU. However, when running on S2C's vu19p, it seems to hang either in _scratch_init or shortly after. The cause is currently unclear.

### Environment

- XiangShan branch: 
- XiangShan commit id: 09d2a32142c64fca898e17c0b943e61ddc331958
- NEMU commit id: 039bf5165d888c10a9436a8d33cdf4cb90b1e7f0
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version: 
- riscv-pk id: de5d22656ca8aa351d0269b688a805b09b30a014

cebarobot, 2024-08-14:
我们目前没有还没有提供用于 FPGA 平台的 openSBI 工作负载的构造教程。您可结合面向软件仿真平台的基于 openSBI 的工作负载构造教程和面向 FPGA 的基于 riscv-pk 的工作负载构造教程来构造您需要的工作负载。

造成内核没有输出或卡死的可能性很多，包括但不限于设备树配置、启动参数、硬件缺陷等，可能还需要进一步观察。

另外，FPGA 平台使用的 UART 模块貌似是 ns16550/8250 兼容的，openSBI 里有对 8250 UART 的支持，大概不需要额外的代码修改。

Currently, there is no tutorial for building openSBI workloads for FPGA platfrom. You could combine tutorial openSBI-Emulation workloads with tutorial riscv-pk-FPGA workloads, to build your workloads.

There are a lot of possible reasons for kernel stucking or not outputing, including wrong device tree, wrong booting parameter, hardware bugs, etc. Futher observation is needed.

By the way, FPGA use ns16550/8250 compatible UART module, which is supported in openSBI. It seems unnecessary to modify the source code.

stwjt, 2024-08-14:
> 我们目前没有还没有提供用于 FPGA 平台的 openSBI 工作负载的构造教程。您可结合面向软件仿真平台的基于 openSBI 的工作负载构造教程和面向 FPGA 的基于 riscv-pk 的工作负载构造教程来构造您需要的工作负载。
>
> 造成内核没有输出或卡死的可能性很多，包括但不限于设备树配置、启动参数、硬件缺陷等，可能还需要进一步观察。
>
> 另外，FPGA 平台使用的 UART 模块貌似是 [...]

好的，感谢你的帮助。
Thanks for your help.

stwjt, 2024-08-20:
> 好的，感谢你的帮助。 Thanks for your help.

感谢帮助 已经找到问题出现在哪里了 进入_scratch_init卡死的原因是没关FW_PIC。。。把FW_PIC=n设置了 然后在把一些不必要的函数注释掉后opensbi就能跑了
Thank you for the assistance. I've identified the cause of the issue that was causing the system to hang during _scratch_init. It was due to FW_PIC not being disabled. After setting FW_PIC to 'n' and commenting out some unnecessary functions, the opensbi can run.