OpenXiangShan/XiangShan #3388: How to debug Xiangshan on FPGA with OpenOCD?

x402, 2024-08-15:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

It is mentioned that [how to use OpenOCD to debug emu of XiangShan](https://xiangshan-doc.readthedocs.io/zh-cn/latest/integration/debug/#connecting-to-debug-module) in the doc, but not XiangShan running on FPGA with JTAG. While we are tring to implement the minimal FPGA system of Nanhu on S2C's Prodigy™ S7-19PD Logic System #3377 ,  I suppose that the CPU crashes on some instruction. So I want to use JTAG and OpenOCD to find out if the CPU stops on some instruction. But we don't know how to begin. So we hope to get some help.
香山文档中有提到[如何在仿真中使用OpenOCD进行调试](https://xiangshan-doc.readthedocs.io/zh-cn/latest/integration/debug/#connecting-to-debug-module)，但是没提到在FPGA平台上运行香山时如何用OpenOCD连接JTAG进行调试。我和同伴正在S2C's Prodigy™ S7-19PD Logic System上搭载南湖的FPGA最小系统进行实验 #3377 ，想知道CPU是否是卡在某个地址的指令无法运行，所以想用JTAG+OpenOCD进行调试，但是现在无从下手，希望可以得到一些指点。

### What did you do before

#3377 

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

cebarobot, 2024-08-19:
We write an tutorial for using FPGA JTAG: [FPGA JTAG - XiangShan-doc](https://docs.xiangshan.cc/zh-cn/latest/integration/fpga/fpga_jtag/). Hope that could help you.

我们撰写了一个 FPGA JTAG 使用的简易教程：[FPGA JTAG - XiangShan-doc](https://docs.xiangshan.cc/zh-cn/latest/integration/fpga/fpga_jtag/)。希望能帮到您。

x402, 2024-08-20:
> We write an tutorial for using FPGA JTAG: [FPGA JTAG - XiangShan-doc](https://docs.xiangshan.cc/zh-cn/latest/integration/fpga/fpga_jtag/). Hope that could help you.
> 
> 我们撰写了一个 FPGA JTAG 使用的简易教程：[FPGA JTAG - XiangShan-doc](https://docs.xiangshan.cc/zh-cn/latest/integration/fpga/fpga_jtag/)。希望能帮到您。

Thank you very much! The tutorial is helpful.
非常感谢，这个教程很有帮助。

Headcrabed, 2024-08-25:
@cebarobot 另外请问有测试过直接使用jlink驱动本身的riscv64调试功能吗？openocd的性能和便利程度比起jlink官方驱动还是差了一些

cebarobot, 2024-08-28:
> @cebarobot 另外请问有测试过直接使用jlink驱动本身的riscv64调试功能吗？openocd的性能和便利程度比起jlink官方驱动还是差了一些

我们没有尝试过，您可以尝试一下。

We haven't try that. You could have a try.

yorange1, 2024-10-09:
@x402 请问你有撰写一些在 S2C's Prodigy™ S7-19PD上运行香山的文档吗？我正在进行一些类似的工作，如果能够得到你们的分享的话将会对我有很大的帮助，十分感谢