OpenXiangShan/XiangShan #3390: wrong address space

mlabaf2, 2024-08-16:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

Hi
I used below instruction to convert .S file to . O
"•	riscv64-linux-gnu-gcc -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -Iriscv-tests/env/p/ -Iriscv-tests/isa/macros/scalar - capstone-ariane/verif/tests/custom/capstone/capcreate.S -o capcreate.o -Wl,--build-id=none"

but after producing .txt file I see wrong address!. 
the code must start from 80000000 but start from 100c0!

[image]

I want to make emu the file on modified XianfShan.

### What did you do before

I want to make emu the file on modified XianfShan.

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

cebarobot, 2024-08-16:
You have to provide an linker script file to linker/compiler, which will tell the linker where to put those sections. 

You could refer to [OpenXiangShan/nexus-am](https://github.com/OpenXiangShan/nexus-am) for details of building baremental applications. There is an linker script example:
* https://github.com/OpenXiangShan/nexus-am/blob/master/am/src/nemu/isa/riscv/boot/loader64.ld
* https://github.com/OpenXiangShan/nexus-am/blob/master/am/src/nemu/ldscript/section.ld
And Makefile:
* https://github.com/OpenXiangShan/nexus-am/blob/28c275b0ea38a341e6ce158d631bb41109863dd2/am/arch/riscv64-xs.mk#L31-L39

mlabaf2, 2024-08-16:
thanks for your help
when I "make ARCH=riscv64-xs" I received below error:
"make: *** No rule to make target '/Makefile.app'.  Stop"
what does "make ARCH=riscv64-xs" exactly do? 

also, what does this sentence means:
"use the ".bin" file as the memory image for RTL simulation"
I create a .bin file from .S assemble of my code. I want to check my .bin file, not the output .bin file of "make ARCH=riscv64-xs" ...

cebarobot, 2024-08-16:
`make ARCH=riscv64-xs` calls `make` (an GNU build tool) to build the software according to the `Makefile` file. You may search "Makefile" on the Internet, and read the `Makefile` file in the repo for more details.

`.bin` file refer to "flat binary files", which means it is just instructions and data without any metadata or file header. This kind of file could just put into memory without any modification. Compiler like gcc usually generate ELF files rather than flat binary files. [This article](https://stackoverflow.com/questions/2427011/what-is-the-difference-between-elf-files-and-bin-files) explains the difference of ELF files and binary files. You may use objcopy to generate binary file from elf file.

Tang-Haojin, 2024-08-16:
You may also specify `AM_HOME` before you `make` in nexus_am.

closed 2025-09-05 by Yan-Muzi