OpenXiangShan/XiangShan #3405: Emu-compile Error

Donkey-Hao, 2024-08-18:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

Hi, I followed the [xs-env](https://xiangshan-doc.readthedocs.io/zh-cn/latest/tools/xsenv-en/) to `make emu CONFIG=MinimalConfig EMU_TRACE=1 -j32`, however I met this problem in the [make-emu.log](https://github.com/user-attachments/files/16647084/make-emu.log) file (part of log). I have checked issues of xs-env and XiangShan, found nothing useful. One more thing, I have tried `gcc/g++ 11.4.0`, faiured.

### What did you do before

I have run  `make riscv64-xs_defconfig` in `/xs-env/NEMU` and  `make verilog` in `/xs-env/XiangShan`. I have run  `make verilog` in `/xs-env/XiangShan` before `make emu`.

### Environment

- XiangShan branch:  master
- XiangShan commit id: 8fae59bba
- NEMU commit id: 3033a69f
- SPIKE commit id: f7d0dba6
- Operating System: 22.04.4 LTS (Jammy Jellyfish)
- gcc version: 11.4.0 & 9.5.0 (*both I have tried*)
- mill version: Mill Build Tool version 0.11.1
- java version: openjdk 11.0.23 2024-04-16

cebarobot, 2024-08-18:
I think this is still your environment problem. 
- On Ubuntu 22.04, the default version of gcc in apt source should be gcc-11. I don't know how you tried gcc-9 and gcc-11 for this step.
- How did you install verilator? Did you use the script [`install-verilator.sh`](https://github.com/OpenXiangShan/xs-env/blob/master/install-verilator.sh) or [`setup-tools.sh`](https://github.com/OpenXiangShan/xs-env/blob/master/setup-tools.sh)?
- Actually, we used clang as default compiler to work with verilator. If you followed the above two script, verilator would use clang as default. 

If possible, I recommand you to try on an brand new Ubuntu 22.04 environment. Also, you could follow the `setup-tools.sh` to complete dependencies and the `install-verilator.sh` to reinstall verilator.

By the way, `make verilog` and `make emu` are different procedure. There is no procedural connection between them.

Donkey-Hao, 2024-08-18:
Hi @cebarobot:
1. For some reason, my default gcc is gcc-9. We could change the gcc version by `update-alternatives --config gcc`, of course we need to set `update-alternatives` first.
2. I have installed verilator before, for XiangShan I changed the version to v4.204. So I don't need to run those scripts to install it.
```
(base) ➜  XiangShan git:(8fae59bba) verilator --version
Verilator 4.204 2021-06-12 rev v4.204
```
3. I will try to change the default cc to clang.

Last but not least, thanks for your help, I will try to install it on a clean OS if step 3 failure.

cebarobot, 2024-08-18:
Hi, thanks for your reply. 

Now XiangShan (master branch) turn to use new version of verilator. As install-verilator.sh says, we currently recommand to use verilator v5.020.

Also, verilator seems to determine the default compiler when building from source code. This feature may inalidate your effort to change the default gcc version. This is just a guess.

By the way, it always takes a lot of time to build an usable environment. 😣

Donkey-Hao, 2024-08-19:
Hi @cebarobot, thanks for your help. I have tackled it with your help, thanks a lot.
I changed the default cc to clang, and config verilator by `./configure CC=clang CXX=clang++ LINK=clang++`. Emu-compile Success!

closed 2024-08-19 by Donkey-Hao