OpenXiangShan/XiangShan #3415: About the clock problem in Xiangshan

LOCKEDGATE, 2024-08-22:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

nanhuv2官网 https://docs.xiangshan.cc/zh-cn/latest/integration/fpga/fpga/ 生成的工程给的实际CPU时钟是50M，然后对应nanhuv2 https://docs.xiangshan.cc/zh-cn/latest/integration/docker/ 给的image里的dts设置的时钟如下图，给的是25M。这两者之间有没有什么联系？如果我想修改实际时钟，假如实际时钟修改为10M，那么dts里需要做相应的修改么？
The actual CPU clock of the project generated by the nanhuv2 official website https://docs.xiangshan.cc/zh-cn/latest/integration/fpga/fpga/ is 50M, and the clock set in the dts of the image given by nanhuv2 https://docs.xiangshan.cc/zh-cn/latest/integration/docker/ is 25M as shown below. Is there any connection between the two? If I want to modify the actual clock, if the actual clock is changed to 10M, do I need to make corresponding changes in the dts?
[image]

### What did you do before

如上

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

cebarobot, 2024-08-28:
十分抱歉回复晚了。

* `clkc` 节点的 `clock-frequency`，应该是指 FPGA 提供的时钟源的频率。
* `clint` 节点的 `clock-frequency-mhz` 指的是 clint 的时钟源的频率，理论上影响 `time` 寄存器的更新频率。
  * 目前这个配置项可能已经废弃，请直接使用 `cpus` 节点的 `timebase-frequency` 项来配置。
  * 在仿真环境下，clint 送入的时钟是在主时钟分频得到的，FPGA 环境下可能需要检查一下具体的代码。
* `cpus` 下面的每个 `cpu` 节点都有一个 `clock-frequency`，它指的是 CPU 实际运行的频率，可以缺省为 0。

相对比较重要的是 `cpus` 节点的 `timebase-frequency`，因为它影响了操作系统计时的准确性。对于较老的 Kernel，也建议同时修改 `clint` 节点的 `clock-frequency-mhz`。

Sorry for late reply.

* `clock-frequency` in node `clkc` is the frequency of FPGA clock.
* `clock-frequency-mhz` in node `clint` is the frequency of clint clock, which influences the frequency of the update of `time` CSR.
  * `clock-frequency-mhz` may be deprecated in latest kernel. Please consider use `timebase-frequency` in node `cpus`.
  * In software emulation environment, clint clock is divided from main clock. You'd better to check the source code.
* `clock-frequency` in every `cpu` node under `cpus` is the frequency of every processor core, and could be default 0.

`timebase-frequency` in node `cpus` is important, which influences the accuracy of Operating System time count. For older kernel, modification of `clock-frequency-mhz` in node `clint` is also required.

LOCKEDGATE, 2024-08-29:
十分感谢，我尝试一下
Thanks a lot ，I'll try it.