OpenXiangShan/XiangShan #3446: How to use Xiangshan tutorial for multi-core testing about spec 2006

LOCKEDGATE, 2024-08-29:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

您好，spec官方教程使用runspec运行spec2006的时候可选-rate来指定多核（线程）测试，我想知道香山给的编译spec2006的教程 https://github.com/OpenXiangShan/CPU2006LiteWrapper/tree/main 能指定在测试的时候开启多核吗？默认是单核测试么
Hello, the official spec tutorial says that when running spec2006 with runspec, you can use -rate to specify multi-core (thread) testing. I would like to know if the spec2006 compilation tutorial provided by Xiangshan https://github.com/OpenXiangShan/CPU2006LiteWrapper/tree/main can specify to enable multi-core during testing? Is it single-core testing by default?

### What did you do before

as above

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

shinezyy, 2024-08-29:
[CPU2006LiteWrapper](https://github.com/OpenXiangShan/CPU2006LiteWrapper/tree/main) only compile binaries according dry-run logs of single-copy SPECCPUs.

To my best knowledges, for some benchmarks of SPECCPU, multi-copy configuration is just running two elfs in parallel. If that satisfies your demand, you can achieve that by modifies the running script specified in initramfs.txt

LOCKEDGATE, 2024-09-02:
Thanks for your reply,I 'll try it.