OpenXiangShan/XiangShan #3530: make verilog 不报错，但无法生成Top.v

han-jianing, 2024-09-10:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

make verilog 不报错，但无法生成Top.v

### What did you do before

执行了make init 和make verilog，内存以及修改小

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

cebarobot, 2024-09-10:
请提供更多信息以协助我们诊断问题，例如：
* `make verilog` 时的输出
* `build` 和 `build/rtl` 目录下的文件列表

Please provide more information to help us diagnose the issue, such as:
* The output when running `make verilog`
* The list of files in the `build` and `build/rtl` directories

han-jianing, 2024-09-10:
[image]
[image]
以上是运行的结果，我第一次使用github提问，有些不熟练，请您多多包涵，谢谢！

cebarobot, 2024-09-10:
@han-jianing 请检查 build 文件夹下的 rtl 文件夹，看看里面有没有你想要的文件。

Please check `rtl` folder under `build`, to see if there is what you wanted.

han-jianing, 2024-09-10:
[image]
这个是我看的一个帖子，但我没有找到XSTop.v文件，rtl下也没有，有一个XSTop.sv文件。这个没关系的吗？麻烦您了，谢谢您！

cebarobot, 2024-09-10:
`build/rtl/XSTop.sv` 就是您要找的 `build/XSTop.v`。这篇文章可能是较早前写的，我们对生成的 Verilog/SystemVerilog 文件位置做了一些修改，不影响正常使用。

`build/rtl/XSTop.sv` is the new name of `build/XSTop.v`. This article may be written earlier. We have made some modifcations to the generated Verilog/SystemVerilog files, which do not affect the use.

han-jianing, 2024-09-10:
好的好的，非常感谢您！