OpenXiangShan/XiangShan #3567: 增加swap后还是报错1 targets failed      xiangshan.test.runMain Subprocess failed

han-jianing, 2024-09-13:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

[image]

### What did you do before

增加了100G SAWP后执行make emu CONFIG=MinimalConfig EMU_TRACE=1 -j8

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

Tang-Haojin, 2024-09-13:
You should first `export NOOP_HOME=$(pwd)`

han-jianing, 2024-09-13:
谢谢，修改好了

---- 回复的原邮件 ----
| 发件人 | Tang ***@***.***> |
| 日期 | 2024年09月13日 18:21 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [OpenXiangShan/XiangShan] 增加swap后还是报错1 targets failed      xiangshan.test.runMain Subprocess failed (Issue #3567) |

You should first export NOOP_HOME=$(pwd)

closed 2024-09-14 by cebarobot