OpenXiangShan/XiangShan #3587: reading fflags changes the status

Donkey-Hao, 2024-09-14:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

In M mode, we set the `mstatus.FS` value then read the `fflags`, we will see the `mstatus` and `sstatus` in Xiangshan changed.
`mstatus = 0x0000000a00006000` and `sstatus = 0x0000000200006000`
[image]

### Expected behavior

`mstatus = 0x0000000a00002000` and `sstatus = 0x0000000200002000`

### To Reproduce

The following zip file contains asm file, elf file, bin file and link file. If you want to assemble and link it, run:
```
riscv64-unknown-elf-as -o "$OBJECT_FILE" "$ASM_FILE"
riscv64-unknown-elf-ld -T "$LINKER_SCRIPT" -o "$NAME" "$OBJECT_FILE"
```
Run Xiangshan with: `./build/emu -i /path/to/readfflags.bin 2>/dev/null`
[read fflags.zip](https://github.com/user-attachments/files/17002122/read.fflags.zip)

### Environment

- XiangShan branch:  master
- XiangShan commit id: 8fae59bba
- NEMU commit id: 3033a69f
- SPIKE commit id: f7d0dba6

### Additional context

I built the env by [xs-env](https://github.com/OpenXiangShan/xs-env).

cebarobot, 2024-09-14:
Your XiangShan and NEMU version may be too old. Please consider to use latest master XiangShan and latest master NEMU to try again. We have solved a lots of bugs since January.

Donkey-Hao, 2024-09-14:
Hi @cebarobot, thanks for your reply. I update the Xiangshan and NEMU, and 're-make' them, now:
```
Xiangshan: 7e1c60713 (HEAD -> master, origin/master, origin/HEAD) fix(vstopi): wrong API usage in InterruptFilter (#3557)
NEMU: 8fba3220 (HEAD -> master, origin/master, origin/HEAD) docs(issues): add issue template (#527)
```
Both two are the latest version. Unfortunately, I got this:
[image]

Those numbers are fixed, I ran several test programs and got same output. You can try the same program I have uploaded.
```
Core 0: ABORT at pc = 0x7e7a8e7a7
instrCnt = 1, cycleCnt = 2,101, IPC = 0.000476
```
Please let me know if I have made any mistakes.
Thanks!

cebarobot, 2024-09-14:
After updating XiangShan (`git pull origin master`), Did you run `make init` to update submodules? You could check this by running `git status` and there should not be anything like this: 
```
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   coupledL2 (new commits)
        modified:   difftest (new commits)
        modified:   ready-to-run (new commits)
        modified:   rocket-chip (new commits)
```

By the way, could you check the NEMU defconfig you used? It should be `riscv64-xs-ref_defconfig`. You could use this defconfig by:
```
make riscv64-xs-ref_defconfig
make -j
```
You could also try the `riscv64-nemu-interpreter-so` as diff ref.

I just run your workloads in my enviroment, where XiangShan is recent version but not the latest. I could run about 14 instrcutions and then traps into endless traps. You could add the following two instructions to end the emulation:
```
mv a0, %0
.word 0x0005006b
```

Donkey-Hao, 2024-09-15:
The problem has been fixed in Xiangshan! I installed the latest version of Xiangshan & NEMU in a brand-new OS. The program runs smoothly without any errors.
Thank you for your patience and help. I really appreciate it.

closed 2024-09-16 by cebarobot