OpenXiangShan/XiangShan #3613: 请问可以提供搭建到VCU128 FPGA开发板的相关文件吗

Satone7, 2024-09-19:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集相关的问题。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the question

我们希望在VCU128 FPGA开发板上搭建香山核，在问题#2214 的[评论](https://github.com/OpenXiangShan/XiangShan/issues/2214#issuecomment-1667508478)中，发现开发组曾经在VCU128搭建过nanhu-G分支。
但是根据香山文档中[FPGA Prototype](https://docs.xiangshan.cc/zh-cn/latest/integration/fpga/fpga/)的指导，没有找到nanhu-G、VCU128相关的tcl脚本等文件，希望能在[env-scripts](https://github.com/OpenXiangShan/env-scripts)中增加nanhu-G搭建到VCU128上的相关脚本和文件，十分感谢！

Tang-Haojin, 2024-09-19:
Thanks for your question. Currently, we do not have related scripts or process specifically for nanhu-G. But you can modify the original scripts to adapt for nanhu-G.

Satone7, 2024-09-20:
> 感谢您的提问。现阶段，我们暂时没有针对nanhu-G的脚本，但是您可以尝试基于原始的脚本针对nanhu-G进行修改。
> 
> Thanks for your question. Currently, we do not have related scripts or process specifically for nanhu-G. But you can modify the original scripts to adapt for nanhu-G.

您好，感谢您的回复。由于我不是专业的FPGA工程师，所以在适配vcu128的工作上遇到了很大的困难。还是不好意思想打扰您，如果贵组曾经有适配过vcu128的相关工作，能否提供一些参考学习的资料，例如针对vcu128进行管脚约束的xdc文件等。任意分支和配置都可以。如果能成功适配vcu128，我乐意参照env-scripts中的格式，整理好针对vcu128的tcl脚本、rtl文件和xdc文件并提交到env-scripts中以供后人参考。再次感谢！

Hello, thank you for your response. Since I am not a professional FPGA engineer, I have encountered significant difficulties in adapting to the VCU128. I apologize for the trouble, but if your team has previously worked on adapting the VCU128, could you kindly provide some reference materials for learning, such as XDC files for pin constraints targeting the VCU128? Any branch or configuration would be helpful. If I manage to successfully adapt the VCU128, I would be happy to follow the format of the env-scripts and organize the TCL scripts, RTL files, and XDC files for the VCU128, then submit them to env-scripts for future reference. Thank you again!

Tang-Haojin, 2024-09-23:
> 您好，感谢您的回复。由于我不是专业的FPGA工程师，所以在适配vcu128的工作上遇到了很大的困难。还是不好意思想打扰您，如果贵组曾经有适配过vcu128的相关工作，能否提供一些参考学习的资料，例如针对vcu128进行管脚约束的xdc文件等。任意分支和配置都可以。如果能成功适配vcu128，我乐意参照env-scripts中的格式，整理好针对vcu128的tcl脚本、rtl文件和xdc文件并提交到 [...]

我们联系了相关开发人员，目前正在整理相关脚本和环境，可能需要一些时间。整理完成后，我们会第一时间通知您。

We've reached out to the developers and they are working on sorting out the scripts and environments. This may take some time. We will notify you as soon as the sorting out is complete.

Satone7, 2024-11-05:
@Tang-Haojin 最近我们这边找到了vu19p的FPGA，于是我也可以直接在vu19p的板子上进行实验了，如果因为我的issue给相关开发人员带来了额外的工作，非常抱歉！

Recently, we have found the FPGA for vu19p here, so I can also conduct experiments directly on the vu19p board. If my issue has caused additional work for the relevant developers, I sincerely apologize!

closed 2024-11-05 by Satone7