OpenXiangShan/XiangShan #3619: How to add new fields to /proc/cpuinfo？

LOCKEDGATE, 2024-09-20:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

我在fpga上启动香山时查看/proc/cpuinfo会看到processor，hart，isa，mmu和uarch字段。如果我想添加新的字段，比如frequency，我应该修改哪个文件？感谢。
When I look at /proc/cpuinfo when I start Xiangshan on my fpga, I see fields for processor, hart, isa, mmu, and uarch. If I want to add a new field, such as frequency, which file should I modify? Thanks.

### What did you do before

as above

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

Tang-Haojin, 2024-09-21:
This is about how Linux kernel gets such info. You may go through Linux kernel source code. `arch/riscv/kernel/cpu.c` and `fs/proc/cpuinfo.c` may be good starting points.

LOCKEDGATE, 2024-09-23:
OK, Thank you very much!

closed 2024-09-23 by LOCKEDGATE