OpenXiangShan/XiangShan #3654: 运行linux.bin对比出错

han-jianing, 2024-09-26:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

运行linux.bin对比出错，但运行
[image]对比成功

hjn@hjn-virtual-machine:~/xs-env/XiangShan$ ./build/emu -i $NOOP_HOME/ready-to-run/linux.bin
emu compiled at Sep 18 2024, 19:18:42
Using simulated 32768B flash
Using simulated 8192MB RAM
The image is /home/hjn/xs-env/XiangShan/ready-to-run/linux.bin
The reference model is /home/hjn/xs-env/NEMU/build/riscv64-nemu-interpreter-so
The first instruction of core 0 has commited. Difftest enabled. 

==============  In the last commit group  ==============
the first commit instr pc of DUT is 0x0000000080009c60
the first commit instr pc of REF is 0x0000000080009c60

============== Commit Group Trace (Core 0) ==============
commit group [00]: pc 0080009c60 cmtcnt 1 <--
commit group [01]: pc 0080009c1a cmtcnt 1
commit group [02]: pc 0080009c1e cmtcnt 1
commit group [03]: pc 0080009c22 cmtcnt 1
commit group [04]: pc 0080009c26 cmtcnt 4
commit group [05]: pc 0080009c32 cmtcnt 1
commit group [06]: pc 0080009c36 cmtcnt 1
commit group [07]: pc 0080009c3a cmtcnt 1
commit group [08]: pc 0080009c3e cmtcnt 1
commit group [09]: pc 0080009c42 cmtcnt 1
commit group [10]: pc 0080009c46 cmtcnt 1
commit group [11]: pc 0080009c4a cmtcnt 1
commit group [12]: pc 0080009c4e cmtcnt 2
commit group [13]: pc 0080009c52 cmtcnt 2
commit group [14]: pc 0080009c58 cmtcnt 1
commit group [15]: pc 0080009c5a cmtcnt 2

============== Commit Instr Trace ==============
[00] commit pc 0000000080009c5a inst 0007b503 wen 1 dst 10 data 000000008000cdd8 idx 010 (02)
[01] commit pc 0000000080009c5c inst 00068733 wen 1 dst 14 data 0000000080046ea0 idx 011
[02] commit pc 0000000080009c60 inst 30551573 wen 1 dst 10 data 0000000080000400 idx 012
[03] exception pc 0000000080009c64 inst 3c0028f3 cause 0000000000000002 <--
[04] commit pc 0000000080009bfc inst 3be898f3 wen 1 dst 17 data 00003ffffffffc00 idx 024
[05] commit pc 0000000080009c00 inst 62b89de3 wen 0 dst 27 data 0000000000000000 idx 025
[06] commit pc 0000000080009c04 inst 0104a503 wen 1 dst 10 data 000000000000000e idx 026 (14)
[07] commit pc 0000000080009c06 inst 00060693 wen 1 dst 13 data 0000000080046ea0 idx 027
[08] commit pc 0000000080009c08 inst f6043c23 wen 0 dst 24 data 0000000000000000 idx 028 (02)
[09] commit pc 0000000080009c0c inst 0015051b wen 1 dst 10 data 000000000000000f idx 029
[10] commit pc 0000000080009c0e inst 00a4a823 wen 0 dst 16 data 0000000000000000 idx 02a (03)
[11] commit pc 0000000080009c10 inst 0007b503 wen 1 dst 10 data 000000008000cdd8 idx 02b (15)
[12] commit pc 0000000080009c12 inst 00068733 wen 1 dst 14 data 0000000080046ea0 idx 02c
[13] commit pc 0000000080009c16 inst 30551573 wen 1 dst 10 data 0000000080000400 idx 02d
[14] commit pc 0000000080009c1a inst 3bf028f3 wen 1 dst 17 data 0000000000000000 idx 02e
[15] commit pc 0000000080009c1e inst 30551073 wen 0 dst 00 data 0000000000000000 idx 02f
[16] commit pc 0000000080009c22 inst f7843683 wen 1 dst 13 data 0000000000000000 idx 000 (16)
[17] commit pc 0000000080009c26 inst 60069ae3 wen 0 dst 21 data 0000000000000000 idx 001
[18] commit pc 0000000080009c2a inst 0007b503 wen 1 dst 10 data 000000008000cdd8 idx 002 (17)
[19] commit pc 0000000080009c2c inst 00060693 wen 1 dst 13 data 0000000080046ea0 idx 003
[20] commit pc 0000000080009c32 inst 30551573 wen 1 dst 10 data 0000000080000400 idx 004
[21] commit pc 0000000080009c36 inst 3bf59073 wen 0 dst 00 data 0000000000000000 idx 005
[22] commit pc 0000000080009c3a inst 30551073 wen 0 dst 00 data 0000000000000000 idx 006
[23] commit pc 0000000080009c3e inst f7843683 wen 1 dst 13 data 0000000000000000 idx 007 (00)
[24] commit pc 0000000080009c42 inst 5e069ce3 wen 0 dst 25 data 0000000000000000 idx 008
[25] commit pc 0000000080009c46 inst 3bf898f3 wen 1 dst 17 data 00003ffffffffc00 idx 009
[26] commit pc 0000000080009c4a inst 5eb898e3 wen 0 dst 17 data 0000000000000000 idx 00a
[27] commit pc 0000000080009c4e inst 0104a503 wen 1 dst 10 data 000000000000000f idx 00b (01)
[... 78 lines cut ...]
Core-0 instrCnt = 894,780, cycleCnt = 583,567, IPC = 1.533294
Seed=0 Guest cycle spent: 583,570 (this will be different from cycleCnt if emu loads a snapshot)
Host time spent: 3,445,692ms

[image]

### What did you do before

[image]
zhe'ge'cheng'g这个成功
[image]
这个对比出错，在NEMU仿真可以通过

非常感谢

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

cebarobot, 2024-09-26:
我们无法根据现有信息判断问题，请在问题描述里补全 Environment 部分，提供 XiangShan、NEMU 和 ready-to-run 的 commit id。

han-jianing, 2024-09-26:
好的谢谢
XiangShan commit id:49162c9ab
NEMU commit id:a6a5f9b1
ready-to-run commit id:3b2f905

han-jianing, 2024-09-26:
XiangShan commit id:49162c9ab
NEMU commit id:a6a5f9b1
ready-to-run commit id:3b2f905
这是我的环境，还需要什么呢

cebarobot, 2024-09-26:
我在较新版本的香山上未能复现这个问题。香山近期在解决大量缺陷，您这个版本的香山有些旧了，您可以更新一下香山和ready-to-run的版本再尝试。

I have tried an new version of XianShan and the bug didn't occur. XiangShan team are solving a lot of bugs thesedays and the version of XiangShan you used is too old. You could try again after updating XiangShan and ready-to-run.

han-jianing, 2024-09-26:
好的，谢谢

---- 回复的原邮件 ----
| 发件人 | Xu, ***@***.***> |
| 日期 | 2024年09月26日 16:58 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [OpenXiangShan/XiangShan] 运行linux.bin对比出错 (Issue #3654) |

我在较新版本的香山上未能复现这个问题。香山近期在解决大量缺陷，您这个版本的香山有些旧了，您可以更新一下香山和ready-to-run的版本再尝试。

I have tried an new version of XianShan and the bug didn't occur. XiangShan team are solving a lot of bugs thesedays and the version of XiangShan you used is too old. You could try again after updating XiangShan and ready-to-run.

han-jianing, 2024-09-27:
我执行了source update-submodule.sh仍然不行，香山怎样更新呢？需要重新git环境吗？

cebarobot, 2024-09-27:
在 `xs-env` 里执行 `source update-submodule.sh` 应该能更新香山和各个子模块。在更新香山之后，需要重新编译香山。

麻烦您提供一下：
* 更新后香山、ready-to-run 的 commit id
* 您编译所用的命令

han-jianing, 2024-09-29:
您好，我已经更新并且重新编译，香山：97e37a223；ready-to-run：8ca5b19
执行命令：
[image]
现在不报错了，出现opensbi字样后不再出现其他信息。

han-jianing, 2024-09-29:
[image]

han-jianing, 2024-09-30:
这个是还在运行吗？好多个小时没有结束正常的吗？感谢解答！国庆节快乐！

cebarobot, 2024-09-30:
> 这个是还在运行吗？好多个小时没有结束正常的吗？感谢解答！国庆节快乐！

这个很难判断。我不太清楚您使用的机器是什么配置，但这个可能在较低配置的机器上可能确实需要很长时间。

ready-to-run 的 linux 大约需要运行 6,000,000 个周期。您可先运行 ready-to-run 下的 microbench 或 coremark，根据它们的周期数（microbench 约 234,000 个周期，coremark 约 300,000 个周期）和实际运行时间来大致估算您运行 linux 所需的时间。

作为参考，在 AMD EPYC 9684X 96-Core Processor 上使用 16 线程的 emu，运行 ready-to-run 里的 linux 大约需要半个小时的时间。更低的 CPU 性能、更少的线程数、虚拟机带来的性能损失都有可能让运行时间变得很长。

同祝您国庆节快乐。

han-jianing, 2024-09-30:
收到，谢谢

---- 回复的原邮件 ----
| 发件人 | Xu, ***@***.***> |
| 日期 | 2024年09月30日 13:34 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [OpenXiangShan/XiangShan] 运行linux.bin对比出错 (Issue #3654) |

这个是还在运行吗？好多个小时没有结束正常的吗？感谢解答！国庆节快乐！

这个很难判断。我不太清楚您使用的机器是什么配置，但这个可能在较低配置的机器上可能确实需要很长时间。

ready-to-run 的 linux 大约需要运行 6,000,000 个周期。您可先运行 ready-to-run 下的 microbench 或 coremark，根据它们的周期数（microbench 约 234,000 个周期，coremark 约 300,000 个周期）和实际运行时间来大致估算您运行 linux 所需的时间。

作为参考，在 AMD EPYC 9684X 96-Core Processor 上使用 16 线程的 emu，运行 ready-to-run 里的 linux 大约需要半个小时的时间。更低的 CPU 性能、更少的线程数、虚拟机带来的性能损失都有可能让运行时间变得很长。

同祝您国庆节快乐。

closed 2025-09-05 by Yan-Muzi