OpenXiangShan/XiangShan #3666: How did Xiangshan achieve 10Mhz uart_16550?

LOCKEDGATE, 2024-09-27:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

如下图所示，uart_16550的axi clock frequency最低支持到25Mhz。但是我也成功跑通了xiangshan 10M的bitfile，我想了解是怎么修改的uart_16550才能够实现让其能够接收低于25M的主时钟？
As shown in the figure below, the minimum supported axi clock frequency of uart_16550 is 25Mhz. But I also successfully ran the xiangshan 10M bitfile. I want to know how to modify uart_16550 to enable it to receive the main clock below 25M?
[image]

### What did you do before

as above

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version:

Tang-Haojin, 2024-09-30:
Excuse me for the late reply. We actually wire a seperate 50MHz clock over UART, which is not affected by core or SoC frequency.

LOCKEDGATE, 2024-10-12:
I am sorry that it took me so long to reply to you. In my original project, I directly used uart 16550, as shown in Figure 1, and it was able to run through at this time. In order to give the uart a separate clock, I found the relevant design of Xiangshan to handle this. I designed the block design according to Xiangshan's method as shown in Figure 2, which only has three modules: axi interconnect, uart 16650 and xlconstant. I instantiated the wrapper generated by the block design in the code, as shown in Figure 3. I connected the pins in Figure 1 to Figure 3. The interfaces of the two are almost the same, except that a few signals such as S00_AXI_0_arprot are not in Figure 1. I assigned all these signals to 0, but Xiangshan seems to have other logic for these pins. The current situation is that I can't run it through according to the design in Figure 3. Is it because my pins are connected incorrectly, or is the value of my signal incorrect? I am looking forward to your reply.
[image]
[image]
[image]

closed 2024-12-05 by LOCKEDGATE