OpenXiangShan/XiangShan #3709: D extension instr `fle.d` bug.

Donkey-Hao, 2024-10-10:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

When we change the `mstatus` by `csrc` instr, then when execute `fle.d`  instr, a bug occurs:
[image]
[image]

**Note:** there are some instructions between `csrc` and `fle.d`, this is the reason why I think this bug is related to `fle.d` instr.

### Expected behavior

When we execute `fle.d`, `mstatus = 0x0000000a00002000`, `sstatus = 0x0000000200002000`.

### To Reproduce

Here is the instr code and bin/elf file to test.
[runxs_files.zip](https://github.com/user-attachments/files/17322806/runxs_files.zip)

### Environment

- xs-env: 2bb84ce
- XiangShan branch: master
- XiangShan commit id: 8bb30a570
- NEMU commit id: 821ea961
- SPIKE commit id: 1.1.1-dev

### Additional context

Xiangshan make: ` make emu CONFIG=DefaultConfig -j48`.
NEMU make: `make clean; make riscv64-xs-ref_defconfig; make -j48`

Tang-Haojin, 2024-10-13:
Thank you for your report. Let me look into this issue.

Tang-Haojin, 2024-10-13:
Caused by inconsistency between XS and NEMU. See https://github.com/OpenXiangShan/NEMU/pull/580.

closed 2024-10-13 by Tang-Haojin