OpenXiangShan/XiangShan #3743: 封装IP

han-jianing, 2024-10-15:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual). 我已经阅读过 RISC-V 指令集手册。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the feature you'd like

能不能把香山的核封装成一个IP？把中断，AXI，JTAG暴露出来呢？

Ma-YX, 2024-10-24:
感谢您的提问。欢迎您进行尝试，为XiangShan开源社区贡献。在尝试过程中如有具体问题，欢迎继续提问，共同探讨。

Thank you for your question. You are welcome to try and contribute to the XiangShan open-source community. If you encounter any specific issues during your attempt, feel free to ask further questions, and we can explore them together.

cebarobot, 2024-10-24:
目前，香山核本身就是包装为 `XSTile` 模块来集成进各种 SoC 系统的，您可以在生成后的 Verilog 中找到 `XSTile` 模块，并参照我们仿真用 SoC 模型的集成方式进行继承。

Currently, XiangShan Core is packaged into module `XSTile` to Integrate into other SoC systems. You could find the `XSTile.sv` in the generated Verilog files and follow the SimTop to integrate it.