OpenXiangShan/XiangShan #3765: make verilog error

yaoyuexiaogege, 2024-10-17:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

make init后make verilog后，如下，出现了错误，请问是什么原因？
mkdir -p build/rtl
time -avp -0 ./build/time.log mill -i xiangshan.runMain top.TopMain
seq-mem --repl-seq-mem-file=XSTop.sv.conf"
--target-dir build/rtl --config DefaultConfig --issue B--firtool-opt "--repl
--num-cores 1 --dump-fir --target systemverilog--split-verilog - firtool-opt
"-O=release - - disable-annotation-unknown - lowering-options=explicitBitcast,disallowLo
calVariables,disallowPortDeclSharing,locationInfoStyle=none" --fpga-platform --disable
-all --remove-assert --reset-gen --disable-always-basic-diff
/scripts/gen_sep_mem.sh "./scripts/vlsi_mem_gen" "buuild/rtl/XSTop.sv.conf" build/rtl
cat: build/rtl/XSTop.sv.conf: No such file or directory
rm: cannot remove 'build/rtl/XSTop.sv.conf.tmp': No such file or directory
make: *** [Makefile:155: build/rtl/XSTop.sv] Erгог 1

### What did you do before

环境已正确配置

### Environment

- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 
- mill version: 
- java version: 

### Additional context

make init后make verilog后，如下，出现了错误，请问是什么原因？
mkdir -p build/rtl
time -avp -0 ./build/time.log mill -i xiangshan.runMain top.TopMain
seq-mem --repl-seq-mem-file=XSTop.sv.conf"
--target-dir build/rtl --config DefaultConfig --issue B--firtool-opt "--repl
--num-cores 1 --dump-fir --target systemverilog--split-verilog - firtool-opt
"-O=release - - disable-annotation-unknown - lowering-options=explicitBitcast,disallowLo
calVariables,disallowPortDeclSharing,locationInfoStyle=none" --fpga-platform --disable
-all --remove-assert --reset-gen --disable-always-basic-diff
/scripts/gen_sep_mem.sh "./scripts/vlsi_mem_gen" "buuild/rtl/XSTop.sv.conf" build/rtl
cat: build/rtl/XSTop.sv.conf: No such file or directory
rm: cannot remove 'build/rtl/XSTop.sv.conf.tmp': No such file or directory
make: *** [Makefile:155: build/rtl/XSTop.sv] Erгог 1

hanyjie, 2024-10-21:
你可以检查一下你的mill有没有安装成功，输入命令`mill --version`

Ma-YX, 2024-10-24:
感谢您的提问。初次部署XiangShan时，请完全按照XiangShan官网文档https://docs.xiangshan.cc/zh-cn/latest/tools/xsenv/中TLDR一节中脚本运行：
```
git clone https://github.com/OpenXiangShan/xs-env
cd xs-env
sudo -s ./setup-tools.sh # use apt to install dependencies, you may modify it to use different pkg manager
source setup.sh # prepare tools, test develop env using a small project
```
您的问题应该在于没有执行source setup.sh导致环境没有配置完全（XiangShan文件夹内为空）

Thank you for your question. When deploying XiangShan for the first time, please follow the steps in the TLDR section of the official XiangShan documentation at https://docs.xiangshan.cc/zh-cn/latest/tools/xsenv/ exactly:
```
git clone https://github.com/OpenXiangShan/xs-env
cd xs-env
sudo -s ./setup-tools.sh # use apt to install dependencies, you may modify it to use different pkg manager
source setup.sh # prepare tools, test develop env using a small project
```
The issue you encountered is likely due to not running source setup.sh, which caused the environment to be improperly configured (the XiangShan folder is empty).

closed 2024-11-05 by Ma-YX