OpenXiangShan/XiangShan #3773: Xiangshan stuck when set `mideleg`

Donkey-Hao, 2024-10-22:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

Run the following test case, xiangshan will stick at `pc = 0x800001a0 : csrs mideleg, a0`. No matter the limited cycle is 25000 or 50000, it stuck at the same pc.

[image]

I thought it maybe a NEMU bug, so I turn off difftest `--no-diff` and run again, stuck at same pc:
[image]

We can see that xiangshan only executed 114 instruction and got stuck. You might think the test program has a bug, so I tested it on NEMU:
[image]

Ignore the NEMU bug, look that NEMU has executed more than 459k instructions! However, xiangshan just executed 114 instructions and stuck.

### Expected behavior

Xiangshan executes smoothly.

### To Reproduce

[stuck.zip](https://github.com/user-attachments/files/17472352/stuck.zip)

### Environment

- XiangShan branch: master
- XiangShan commit id: e11ec86cc
- NEMU commit id: 821ea961
- SPIKE commit id: 1.1.1-dev

### Additional context

I recommend you check vcd file, I did not log it because of its large scale.

cebarobot, 2024-10-24:
Thanks for your report. I'm dealing with this.

lewislzh, 2024-10-24:
Firstly, NEMU did not reach the point where XiangShan previously encountered the stcuk. You can enable the debug option to check this. Since NEMU's configuration has CONFIG_REPORT_ILLEGAL_INSTR=y enabled, it will abort directly when an illegal instruction exception occurs. As for the instruction count displayed by NEMU, it is not accurate after enabling opt_perf.

Secondly, regarding the XiangShan stuck issue, your test includes a wfi instruction. When NEMU and XiangShan execute this instruction during the diff process, they will wait for an interrupt or exceed 2^20 cycles. Therefore, the 'stuck' phenomenon you mentioned is normal.
[image]

Finally, after disabling the CONFIG_REPORT_ILLEGAL_INSTR option in NEMU, we ran your test using NEMU alone. Since NEMU treats the wfi instruction as a noop when handling it independently, it does not result in the same stuck behavior as XiangShan. However, in your test, an ecall instruction is added in the finish section, and the exception handling directs ecall to write_to_host, causing a re-execution, which leads to an infinite loop without exiting. I'm not sure if this is the behavior you intended
[image]
[image]

Tang-Haojin, 2024-10-24:
> Secondly, I have a log file, [xsdifftest.log](https://github.com/user-attachments/files/17507527/xsdifftest.log). Sorry, I did not add this file before because I did not set difftest correctly to trace instr until afternoon. We can see that Xiangshan did not execute `wfi` instruction, and we can see where it stuck. Please check it, I hope it will help you.

XS Do reach `wfi` after the last `mret` instruction because `.insn 4, 0x700b` at 0x800001c8 seems to be an illegal instruction and in trap handler, it makes mepc + 4 and then `mret`, which makes XS reach `wfi` instruction at 0x800001cc.

Donkey-Hao, 2024-10-24:
Ah, I see, thank you all!

closed 2024-10-24 by Donkey-Hao