OpenXiangShan/XiangShan #3782: 生成比特流

han-jianing, 2024-10-24:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

执行make bitstream CORE_DIR=$NOOP_HOME/build后，等待几分钟终端就结束了，没有报错，应该也没有生成比特流

### What did you do before

git checkout 09d2a32142c64fca898e17c0b943e61ddc331958
make init
make clean
make verilog
git clone https://github.com/OpenXiangShan/env-scripts.git
cd xs_nanhu_fpga
make update_core_flist CORE_DIR=$NOOP_HOME/build
make nanhu CORE_DIR=$NOOP_HOME/build
make bitstream CORE_DIR=$NOOP_HOME/build

### Environment

- gcc version: 11.4.0
- mill version: 0.11.8
- java version: 11.0.24

### Additional context

[image]

poemonsense, 2024-10-24:
As the log shows (in the last 5 lines), please first check the log file for impl_1 at `/home/hjn/xs-env/env-scripts/xs_nanhu_fpga/xs_nanhu/xs_nanhu.runs/impl_1/runme.log`

cebarobot, 2024-10-24:
请确认一下您 `$NOOP_HOME` 环境变量是否有正确设置？运行 `echo $NOOP_HOME`，检查其是否为 XiangShan 文件夹的路径。

Please check your environment variable `$NOOP_HOME`. Run `echo $NOOP_HOME` and check whether the result is your XiangShan dir.

han-jianing, 2024-10-24:
您好，是正确的，/home/xs-env/Xiangshan

---- 回复的原邮件 ----
| 发件人 | Xu, ***@***.***> |
| 日期 | 2024年10月24日 16:25 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [OpenXiangShan/XiangShan] problem about generating bitstream (Issue #3782) |

请确认一下您 $NOOP_HOME 环境变量是否有正确设置？运行 echo $NOOP_HOME，检查其是否为 XiangShan 文件夹的路径。

Please check your environment variable $NOOP_HOME. Run echo $NOOP_HOME and check whether the result is your XiangShan dir.

han-jianing, 2024-10-24:
您好，是正确的，/home/xs-env/XiangShan

---- 回复的原邮件 ----
| 发件人 | Xu, ***@***.***> |
| 日期 | 2024年10月24日 16:25 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [OpenXiangShan/XiangShan] problem about generating bitstream (Issue #3782) |

请确认一下您 $NOOP_HOME 环境变量是否有正确设置？运行 echo $NOOP_HOME，检查其是否为 XiangShan 文件夹的路径。

Please check your environment variable $NOOP_HOME. Run echo $NOOP_HOME and check whether the result is your XiangShan dir.

xianfeili, 2024-10-24:
> 终端结束之后，还没有生成这个文件，是终端退出了但比特流还在生成中嘛？
> […](#)

1. ps命令看看是否还有vivado的进程在运行

han-jianing, 2024-10-24:
没有在运行

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年10月24日 16:37 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [OpenXiangShan/XiangShan] problem about generating bitstream (Issue #3782) |

终端结束之后，还没有生成这个文件，是终端退出了但比特流还在生成中嘛？
…
---- Replied Message ---- | From | Yinan @.> | | Date | 10/24/2024 14:42 | | To | @.> | | Cc | @.>@.> | | Subject | Re: [OpenXiangShan/XiangShan] problem about generating bitstream (Issue #3782) | As the log shows (in the last 5 lines), please first check the log file for impl_1 at /home/hjn/xs-env/env-scripts/xs_nanhu_fpga/xs_nanhu/xs_nanhu.runs/impl_1/runme.log

ps命令看看是否还有vivado的进程在运行

xianfeili, 2024-10-24:
> 没有在运行
> […](#)

那就用vivado的GUI界面打开该工程，看看是在运行还是已经报错，如果有报错可以把相关message贴出来

han-jianing, 2024-10-24:
好的，谢谢，我先看一下，解决不了再咨询您，衷心感谢

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年10月24日 16:57 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [OpenXiangShan/XiangShan] problem about generating bitstream (Issue #3782) |

没有在运行
…
---- 回复的原邮件 ---- | 发件人 | @.> | | 日期 | 2024年10月24日 16:37 | | 收件人 | @.> | | 抄送至 | @.>@.> | | 主题 | Re: [OpenXiangShan/XiangShan] problem about generating bitstream (Issue #3782) | 终端结束之后，还没有生成这个文件，是终端退出了但比特流还在生成中嘛？ … ---- Replied Message ---- | From | Yinan @.> | | Date | 10/24/2024 14:42 | | To | @.> | | Cc | @.>@.> | | Subject | Re: [OpenXiangShan/XiangShan] problem about generating bitstream (Issue #3782) | As the log shows (in the last 5 lines), please first check the log file for impl_1 at /home/hjn/xs-env/env-scripts/xs_nanhu_fpga/xs_nanhu/xs_nanhu.runs/impl_1/runme.log

那就用vivado的GUI界面打开该工程，看看是在运行还是已经报错，如果有报错可以把相关message贴出来

han-jianing, 2024-10-25:
查看了一下，似乎是XiangShan/build中生成的XSTop.sv的端口名称，与env_scripts下，XSTop_wrapper.v的端口不匹配，例如后者为memory_0_awready，但前者为memory_awready。这是我XiangShan/build生成的rtl有问题嘛？还是说需要手动修改呢？麻烦您了！

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年10月24日 16:57 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [OpenXiangShan/XiangShan] problem about generating bitstream (Issue #3782) |

没有在运行
…
---- 回复的原邮件 ---- | 发件人 | @.> | | 日期 | 2024年10月24日 16:37 | | 收件人 | @.> | | 抄送至 | @.>@.> | | 主题 | Re: [OpenXiangShan/XiangShan] problem about generating bitstream (Issue #3782) | 终端结束之后，还没有生成这个文件，是终端退出了但比特流还在生成中嘛？ … ---- Replied Message ---- | From | Yinan @.> | | Date | 10/24/2024 14:42 | | To | @.> | | Cc | @.>@.> | | Subject | Re: [OpenXiangShan/XiangShan] problem about generating bitstream (Issue #3782) | As the log shows (in the last 5 lines), please first check the log file for impl_1 at /home/hjn/xs-env/env-scripts/xs_nanhu_fpga/xs_nanhu/xs_nanhu.runs/impl_1/runme.log

那就用vivado的GUI界面打开该工程，看看是在运行还是已经报错，如果有报错可以把相关message贴出来

Tang-Haojin, 2024-10-25:
> 查看了一下，似乎是XiangShan/build中生成的XSTop.sv的端口名称，与env_scripts下，XSTop_wrapper.v的端口不匹配，例如后者为memory_0_awready，但前者为memory_awready。这是我XiangShan/build生成的rtl有问题嘛？还是说需要手动修改呢？麻烦您了！

请您再次确认一下您的香山版本是否使用的是南湖 `09d2a32`. 该版本生成的顶层文件是 `build/XSTop.v` 而非 `build/rtl/XSTop.sv`. 您可使用 `git log` 来检查您的版本，并用 `git status` 来检查是否有额外的更改。

如果可能，请您尝试重新 clone 并再次 checkout 到该版本，然后再次尝试文档给出的步骤。

Please check your XiangShan commit is `09d2a32`. This version generated the top file `build/XSTop.v` rather than `build/rtl/XSTop.sv`. You could use `git log` to check the commit id and use `git status` to check uncommitted changes.

If possible, please try to re-clone the repo and checkout that commit, and then try the process again.

han-jianing, 2024-10-25:
好的，我试一下，谢谢！

---- 回复的原邮件 ----
| 发件人 | Tang ***@***.***> |
| 日期 | 2024年10月25日 12:44 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [OpenXiangShan/XiangShan] problem about generating bitstream (Issue #3782) |

查看了一下，似乎是XiangShan/build中生成的XSTop.sv的端口名称，与env_scripts下，XSTop_wrapper.v的端口不匹配，例如后者为memory_0_awready，但前者为memory_awready。这是我XiangShan/build生成的rtl有问题嘛？还是说需要手动修改呢？麻烦您了！

请您再次确认一下您的香山版本是否使用的是南湖 09d2a32. 该版本生成的顶层文件是 build/XSTop.v 而非 build/rtl/XSTop.sv. 您可使用 git log 来检查您的版本，并用 git status 来检查是否有额外的更改。

如果可能，请您尝试重新 clone 并再次 checkout 到该版本，然后再次尝试文档给出的步骤。

Please check your XiangShan commit is 09d2a32. This version generated the top file build/XSTop.v rather than build/rtl/XSTop.sv. You could use git log to check the commit id and use git status to check uncommitted changes.

If possible, please try to re-clone the repo and checkout that commit, and then try the process again.

han-jianing, 2024-10-25:
麻烦问一下，单核不能使用这个仓库嘛？，我make verilog时没有指定NUM_CORE=2

---- 回复的原邮件 ----
| 发件人 | Tang ***@***.***> |
| 日期 | 2024年10月25日 12:44 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [OpenXiangShan/XiangShan] problem about generating bitstream (Issue #3782) |

查看了一下，似乎是XiangShan/build中生成的XSTop.sv的端口名称，与env_scripts下，XSTop_wrapper.v的端口不匹配，例如后者为memory_0_awready，但前者为memory_awready。这是我XiangShan/build生成的rtl有问题嘛？还是说需要手动修改呢？麻烦您了！

请您再次确认一下您的香山版本是否使用的是南湖 09d2a32. 该版本生成的顶层文件是 build/XSTop.v 而非 build/rtl/XSTop.sv. 您可使用 git log 来检查您的版本，并用 git status 来检查是否有额外的更改。

如果可能，请您尝试重新 clone 并再次 checkout 到该版本，然后再次尝试文档给出的步骤。

Please check your XiangShan commit is 09d2a32. This version generated the top file build/XSTop.v rather than build/rtl/XSTop.sv. You could use git log to check the commit id and use git status to check uncommitted changes.

If possible, please try to re-clone the repo and checkout that commit, and then try the process again.

xianfeili, 2024-10-25:
> 麻烦问一下，单核不能使用这个仓库嘛？，我make verilog时没有指定NUM_CORE=2
> […](#)

单核也是这个仓库的

han-jianing, 2024-10-30:
抱歉再次打扰。
我现在生成的XSTop.v文件是这样的，里面没有XSTop模块。

//commit 09d2a32142c64fca898e17c0b943e61ddc331958
//Author: Yinan Xu <xuyinan@ict.ac.cn>
//Date:   Sun Feb 5 10:43:57 2023 +0800
//
//    fix sramClkDivBy2
//diff --git a/build.sc b/build.sc
//index 2fcec11bd..c5bf50423 100644
//--- a/build.sc
//+++ b/build.sc
//@@ -134,7 +134,7 @@ trait CommonXiangShan extends XSModule with SbtModule { m =>
// 
//   override def millSourcePath = os.pwd
// 
//-  override def forkArgs = Seq("-Xmx64G", "-Xss256m")
//+  override def forkArgs = Seq("-Xmx24G", "-Xss256m")
// 
//   override def ivyDeps = super.ivyDeps() ++ Seq(ivys.chiseltest)
// 
module BankBinder(
  output         auto_in_a_ready,
  input          auto_in_a_valid,
  input  [2:0]   auto_in_a_bits_opcode,
  input  [2:0]   auto_in_a_bits_param,
  input  [2:0]   auto_in_a_bits_size,
  input  [3:0]   auto_in_a_bits_source,
  input  [35:0]  auto_in_a_bits_address,
  input  [31:0]  auto_in_a_bits_mask,
  input  [255:0] auto_in_a_bits_data,

han-jianing, 2024-10-30:
已经解决，非常感谢！

closed 2024-10-30 by han-jianing