OpenXiangShan/XiangShan #3829: `mcause` error in difftest when `raise intr cause NO: 4`

Donkey-Hao, 2024-11-02:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

1. Run the workload, difftest shows (complete log in the log file):
```
privilegeMode: 3
 mcause different at pc = 0x0080000b80, right= 0x0000000000000004, wrong = 0x0000000000000005
Core 0: ABORT at pc = 0x80000bae
Core-0 instrCnt = 495, cycleCnt = 5,659, IPC = 0.087471
```

2. A trap is `raise intr cause NO: 4, epc: 8000e946`, NO.4 is `EX_LAM, // load address misaligned`. However, spike did not get into the trap, detail please see the tail of log file.
```
0x0000000080000b98:   9b 12 e3 01                 slliw   t0,t1,30
[src/cpu/cpu-exec.c:52,debug_hook] 0x0000000080000b9c:   67 83 04 00                 jalr   t1,s1,0
0x0000000080000b9c:   67 83 04 00                 jalr   t1,s1,0
[src/isa/riscv64/system/intr.c:69,raise_intr] raise intr cause NO: 4, epc: 8000e946
==============  In the last commit group  ==============
the first commit instr pc of DUT is 0x0000000080000b80
the first commit instr pc of REF is 0x0000000080000b80
============== Commit Group Trace (Core 0) ==============
```

### Expected behavior

I have run spike to test it, however the `mcause` value in spike is different from NEMU and Xiangshan. Part of spike log in the end of log file. I do not know whether Xiangshan is right or NEMU is right or spike, because Xiangshan supports some extensions, but spike doesn't.

### To Reproduce

[218.zip](https://github.com/user-attachments/files/17607340/218.zip)

### Environment

- XiangShan branch: master
- XiangShan commit id: 7af39ad2d
- NEMU commit id: 9ccee25c
- SPIKE commit id: 1.1.1-dev

### Additional context

Run spike with: `spike --pc=0x80000000 --isa=rv64gc -d xxx.elf`

NewPaulWalker, 2024-11-04:
Thank you for your issue. This is a bug in NEMU, as it checks for misalignment of an MMIO address before confirming that the address is indeed an MMIO address. I have fixed it in [NEMU](https://github.com/OpenXiangShan/NEMU/pull/628)

Donkey-Hao, 2024-11-04:
Thanks for your confirmation. 
I have a question, how can you make sure this is a bug of NEMU not xiangshan. I want to learn about it.

NewPaulWalker, 2024-11-04:
> Thanks for your confirmation. I have a question, how can you make sure this is a bug of NEMU not xiangshan. I want to learn about it.

For your test, there is an inconsistency between XiangShan and NEMU on a fld instruction.
NEMU raised LAM, while XiangShan raised LAF.
Additionally, both XiangShan and NEMU support regular misaligned load/store instructions, so we do not raise LAM in this case, which indicates that this is a bug in NEMU.

In addition, I observed that the instruction flow executed in XiangShan and NEMU differs from the instruction flow you provided from Spike.
[image]

During the differential testing between XiangShan and NEMU, the value of register s1 was 0x8000e947, and after executing jalr, the PC was updated to 0x8000e946, which is not an expected PC.
After disassembling the ELF file you provided, there are only two instructions with PCs at 0x8000e944 and 0x8000e948.
I am unsure whether you intended to generate such a test.
The next instruction at PC 0x8000e946 is an fld instruction under the C extension.
However, the instruction flow you provided from Spike jumps to the PC at 0x8000e940.
[image]

This is also why Spike did not raise an exception during this period and the mcause retains its old value.

By the way, you can also use Spike as a reference for differential testing. We support differential testing between XiangShan and NEMU, XiangShan and Spike, as well as NEMU and Spike. The method used is similar.

Donkey-Hao, 2024-11-04:
Thanks for your patient and detailed reply. I still have a lot of questions.
> NEMU raised LAM, while XiangShan raised LAF.
Additionally, both XiangShan and NEMU support regular misaligned load/store instructions, so we do not raise LAM in this case, which indicates that this is a bug in NEMU.

> During the differential testing between XiangShan and NEMU, the value of register s1 was 0x8000e947, and after executing jalr, the PC was updated to 0x8000e946, which is not an expected PC.

1. Here both XiangShan & NEMU jumped to `pc = 0x8000e946`, and they both tried to execute the "fld instruction under C extension", if we suppose they did not raise tarp, they executed `fld` successfully, what is the expected behavior when `PC = 0x8000e946 + 4`?

> This is also why Spike did not raise an exception during this period and the mcause retains its old value.

2. As the picture shows: spike execute the instruction at 0x8000e940 `lui   t1, 0x80028` after pc = 0x80000b9c `jarl`, not 0x8000e946 `fld`, so I am confused why `pc = 0x8000e946` in XiangShan and NEMU, but spike the `pc = 0x8000e940` ? The instruction trace in spike is different from XiangShan, which one is right?

Thanks for your time. Thank you a lot. 
Best wishes.

NewPaulWalker, 2024-11-04:
> 1. Here both XiangShan & NEMU jumped to `pc = 0x8000e946`, and they both tried to execute the "fld instruction under C extension", if we suppose they did not raise tarp, they executed `fld` successfully, what is the expected behavior when `PC = 0x8000e946 + 4`?

When PC = 0x8000e946, it is a fld instruction under the C extension, which means this instruction is only two bytes long.
Therefore, the address of the next instruction should be PC = 0x8000e946 + 2, not + 4.
For the test you provided, both XiangShan and NEMU would continue executing the next instruction, which is the blt instruction at PC = 0x8000e948.

If the fld instruction is not a C extension instruction, it should consist of 4 bytes, specifically at addresses 0x8000e946, 0x8000e947, 0x8000e948, and 0x8000e949.
If it can be successfully executed, the next instruction's PC should be 0x8000e94a. And execute the corresponding instruction.

If you still have questions about above, you should first read the relevant sections on the C extension in the RISC-V manual.

NewPaulWalker, 2024-11-04:
> 2\. As the picture shows: spike execute the instruction at 0x8000e940 `lui   t1, 0x80028` after pc = 0x80000b9c `jarl`, not 0x8000e946 `fld`, so I am confused why `pc = 0x8000e946` in XiangShan and [...]

You should analyze which instruction caused the value 0x8000e947 to appear in the s1 register in XiangShan and NEMU.
Alternatively, you can use XiangShan and Spike for differential testing, rather than running them separately and then comparing their differences.

Donkey-Hao, 2024-11-04:
> When PC = 0x8000e946, it is a fld instruction under the C extension, which means this instruction is only two bytes long. Therefore, the address of the next instruction should be PC = 0x8000e946 + [...]

Thanks for your patient and kind reply. I will check Manual.

NewPaulWalker, 2024-11-04:
> > Thank you for your issue. This is a bug in NEMU, as it checks for misalignment of an MMIO address before confirming that the address is indeed an MMIO address. I have fixed it in [NEMU](https://github.com/OpenXiangShan/NEMU/pull/628)
> 
> I previously encountered a similar issue with both `fld` and `sw` instructions. After reviewing your fix, it seems that only load-related cases were addressed, while store cases may still need attention. Could you please take a look at this? For more details, please see: [OpenXiangShan/NEMU#629](https://github.com/OpenXiangShan/NEMU/issues/629)
> 
> Thank you very much for your help!

Thank you again for your reminder.

closed 2024-11-04 by huxuan0307