OpenXiangShan/XiangShan #3830: [Bug Report] Load access fault and store_address_misaligned cause processor to deadlock

chenhychen, 2024-11-04:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

The test cases with instructions triggering a load access fault exception or store address misaligned exception impede the processor's ability to commit instructions when using NEMU for differential testing. 
[image]

### Expected behavior

I have tested the same test cases on Spike, which throws exceptions correctly.
[image]

### To Reproduce

I have attached the source codes that trigger this issue. 
I use VCS for simulation. example execution command: `./build/simv +workload=noissue.bin +diff=./xs-env/XiangShan/ready-to-run/riscv64-nemu-interpreter-so`

[xs_bug.zip](https://github.com/user-attachments/files/17613075/xs_bug.zip)

### Environment

I installed XiangShan from https://github.com/OpenXiangShan/xs-env `2bb84ce`
- XiangShan commit id: 49162c9
- NEMU commit id: a6a5f9b
- SPIKE commit id: 530af85 (tag v1.1.0)

Ma-YX, 2024-11-04:
Thank you for your question. If possible, please provide the command you used to compile XiangShan.
 We compile XiangShan (vcs) using the following commands: ```make simv ENABLE_XPROP=1``` (for the TileLink version, corresponding to DefaultConfig) or ```make simv CONFIG=KunminghuV2Config ENABLE_XPROP=1``` (for the CHI version). When running the test using ```./build/simv +workload=$(PATH_TO_WORKLOAD) +diff=$(NEMU_HOME)/ready-to-run/riscv64-nemu-interpreter-so``` or use ```+diff=no-diff``` option, we could not reproduce the issue you mentioned, i.e., we did not observe the processor not committing the first instruction. You can have another try with commands provided above.
Additionally, the current version of XiangShan supports unaligned memory accesses except for MMIO, AMO, and RVV. You need to enable unaligned access in Spike for proper comparison with XiangShan/Nemu.

chenhychen, 2024-11-04:
Hi @Ma-YX ,

Thank you for your quick responses! I will try on my side.

Spike provides a flag `--enable-misaligned` to support unaligned memory accesses.
Is there anything else I should modify the Spike?

Ma-YX, 2024-11-05:
The --enable-misaligned option in Spike is a global setting, whereas currently XiangShan does not support misaligned address access for MMIO, AMO, and RVV, creating a difference between the two.

 Additionally, it is recommended to use Spike for differential testing rather than running Spike standalone.

chenhychen, 2024-11-05:
Thank you. I will check it on my side.

closed 2024-11-05 by chenhychen