OpenXiangShan/XiangShan #3834: Instruction trace different between xiangshan & spike

Donkey-Hao, 2024-11-04:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

1. After `80000b9c:	00048367          	jalr	t1,s1`, **spike `pc = 0x8000e940`**, but Xiangshan `raise intr cause NO: 4, epc: 8000e946`, clearly **Xiangshan `pc = 0x8000e946`** which **is different from spike**. Is it a bug of Xiangshan?

[image]
[image]

2. During showing instruction trace, some **pc was not shown** by Xiangshan or NEMU, but spike showed those pc. **Is this is a flaw of difftest?**

[image]
[image]

### What did you do before

I did difftest: #3829  
[218.zip](https://github.com/user-attachments/files/17618078/218.zip)

### Environment

- XiangShan branch: master
- XiangShan commit id: [7af39ad2](https://github.com/OpenXiangShan/XiangShan/commit/7af39ad2ddb1305b2c4ddf4c3a9663a7c3615fa6)
- NEMU commit id:  9ccee25c
- SPIKE commit id: 1.1.1-dev
- Operating System: Linux
- gcc version: 9.5.0
- mill version:  0.11.6
- java version: openjdk 11.0.23

NewPaulWalker, 2024-11-05:
> 1. After `80000b9c:	00048367          	jalr	t1,s1`, **spike `pc = 0x8000e940`**, but Xiangshan `raise intr cause NO: 4, epc: 8000e946`, clearly **Xiangshan `pc = 0x8000e946`** which **is different from spike**. Is it a bug of Xiangshan?

I don’t think this is a bug in XiangShan.
I performed a differential test using XiangShan and Spike and checked the results: the instruction flow remains consistent, and after the jalr instruction at PC = 0x80000b9c, the PC is updated to 0x8000e946.
[image]

As I mentioned before, I **strongly** recommend that you use differential testing to compare the behavior between DUT (e.g. XiangShan) and REF (e.g. Spike), rather than running XiangShan and Spike separately and then comparing the results.
This is because you cannot guarantee that the architectural state of XiangShan and Spike will be consistent after reset or after executing each instruction.
Although both can execute instructions individually, any differences in the instruction flow or some results may only appear after a lot of instructions, making it difficult to find the cause of the discrepancy.

NewPaulWalker, 2024-11-05:
> 2\. During showing instruction trace, some **pc was not shown** by Xiangshan or NEMU, but spike showed those pc. **Is this is a flaw of difftest?**

This is not a flaw of difftest.
We do not currently support comparison of certain CSR registers between XiangShan and NEMU.
The log you provided shows the instruction flow sequence from NEMU, but it contains reads and writes to the mcycle and minstret CSRs (via the csrrw instruction).
You can check the code snippets from [XiangShan](https://github.com/OpenXiangShan/XiangShan/blob/master/src/main/scala/xiangshan/backend/fu/NewCSR/NewCSR.scala#L817)—when accessing such registers, NEMU will choose to skip them and instead copy the architectural state from XiangShan after that instruction has been executed.
As a result, these two instructions are not actually executed in NEMU until the next instruction that NEMU can execute, which is the `csrw mcause, zero` instruction at `PC=0x8000003c`. Therefore, these two instructions are missing from the log of the instruction sequence.

Donkey-Hao, 2024-11-06:
Thanks for your reply. 

https://github.com/OpenXiangShan/XiangShan/issues/3834#issuecomment-2456142133 You tried in difftest with spike, I appreciate what you have done. 
I don't necessarily need XiangShan and Spike to be consistent. I know that XiangShan supports certain features that Spike does not. So, what is the **_root causes_** of the problem? I am just curiosity about this problem, so the **tag is problem** not bug. 

Sincerely.

Donkey-Hao, 2024-11-06:
> This is not a flaw of difftest. We do not currently support comparison of certain CSR registers between XiangShan and NEMU. The log you provided shows the instruction flow sequence from NEMU, but it [...]

It seems that I can change the code of NEMU to log those instructions.
Thanks very much!

NewPaulWalker, 2024-11-06:
> Thanks for your reply.
> 
> [#3834 (comment)](https://github.com/OpenXiangShan/XiangShan/issues/3834#issuecomment-2456142133) You tried in difftest with spike, I appreciate what you have done. I don't necessarily need XiangShan and Spike to be consistent. I know that XiangShan supports certain features that Spike does not. So, what is the **_root causes_** of the problem? I am just curiosity about this problem, so the **tag is problem** not bug.
> 
> Sincerely.

I believe the root cause of the issue is that XiangShan and Spike are not fully consistent.
As for the specific instruction where the inconsistency occurs, unfortunately, I can't tell you without using differential testing.
If you're interested, you should try to figure it out yourself.
For example, why is the value in the s1 register inconsistent?
Which instruction causes the discrepancy to start?
Additionally, I can provide one more piece of information: after reset, the PC in XiangShan is 0x10000000, and then it executes three instructions that jump to 0x80000000 to run your workload, using the t0 register.
I'm not sure if this might cause the inconsistency between XiangShan and Spike in your test.
Good luck!

closed 2024-11-08 by NewPaulWalker