OpenXiangShan/XiangShan #3860: Wrong `mstatus, mtval` value when Xiangshan executes illegal instruction.

ha0lyu, 2024-11-11:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

When xiangshan execute an illegal instruction `fld     fa1, 160(a1)`, xiangshan gets wrong `mstatus & mtval` comparing with `ready-to-run/spike-so`, `mtval` is certainly wrong because it is also different from the result of NEMU. Please check the two log files for details.

### Expected behavior

mstatus & mtval is right value.

### To Reproduce

[test.zip](https://github.com/user-attachments/files/17702527/test.zip)

### Environment

- XiangShan branch: master
- XiangShan commit id: 4376b5255
- NEMU commit id: a4815f99
- SPIKE commit id: 1.1.0-dev

ha0lyu, 2024-11-11:
`mstatus = 0x0000040a00001800` in Xiangshan & NEMU, it seems that the [spike-so](https://github.com/OpenXiangShan/riscv-isa-sim) has a bug? Please check whether it is a bug of the  [spike-so](https://github.com/OpenXiangShan/riscv-isa-sim).

ha0lyu, 2024-11-11:
Hi @fly-1011,
Thanks for your kindly reminder. Yes, this may be a same bug as you mentioned before. I didn't know that issue because I just checked issues of Xiangshan. Next time, please issue it in Xiangshan again.

> Regarding the discrepancy you mentioned with the mstatus register, it may be due to the fact that both XiangShan and NEMU have implemented the SMDBLTRP extension (RISC-V M-mode Double Trap Extension), which is not implemented in Spike.

As for the error `mstatus` value, it seems that this is a bug in [spike of DiffTest](https://github.com/OpenXiangShan/riscv-isa-sim), which is a tool belongs to OpenXiangShan not the [riscv-isa-sim: spike](https://github.com/riscv-software-src/riscv-isa-sim).

Let's keep this issue open until this bug fixed.

NewPaulWalker, 2024-11-12:
Thanks for @fly-1011 response.
The mismatch in `mtval` is the same issue as https://github.com/OpenXiangShan/NEMU/issues/648.
As for the mismatch in `mstatus` between XiangShan and Spike, as I mentioned in https://github.com/OpenXiangShan/NEMU/issues/641#issuecomment-2459041075, it is because Spike has not implemented the SMDBLTRP extension yet.
Also, we do not currently plan to implement this extension on Spike.

lewislzh, 2024-11-12:
For `sm/ssbltrp`, there are some implementation-defined inconsistencies between Xiangshan and Spike. Therefore, the relevant extensions are not enabled for `spike-so`. Additionally, since `smdbltrp` is based on the `smrnmi` extension, which is also mostly vestigial in Spike [a5fdc4efbf9f79a890fe5d1c144880548b011d90](https://github.com/OpenXiangShan/riscv-isa-sim/commit/a5fdc4efbf9f79a890fe5d1c144880548b011d90), we currently see no need to conduct a diff for the above extensions with Spike. In the future, there may be plans to align the behavior of Spike with Xiangshan. If you’re interested, you are welcome to submit a PR to the Xiangshan Spike repository.

lewislzh, 2024-11-14:
I have completed the implementation of the SMRNMI extension and the dbltrp-related extension under XiangShan’s Spike [3870 ](https://github.com/OpenXiangShan/XiangShan/pull/3870 ), [54](https://github.com/OpenXiangShan/riscv-isa-sim/pull/54), [658](https://github.com/OpenXiangShan/NEMU/pull/658) . There should be no mstatus.mdt-related issues in future Spike diffs.

ha0lyu, 2024-11-14:
Hi @lewislzh,
I really appreciate what you have done.
I have tested it in **the newest version**, it works! No more `mstatus` different for now.
However, this test case raise `mcause` different:
```
sh      s2, 1259(s0)   # test case
-------------------
privilegeMode: 3
 mcause different at pc = 0x0080000030, right= 0x0000000000000007, wrong = 0x0000000000000006
```
I ran this test case, the `mcause` value is 6 in the official spike, which is the same as XiangShan. But in spike-diff, it is 7, nemu-diff reported nothing.

NewPaulWalker, 2024-11-15:
This seems to be a new issue. Please create a new issue for it.

closed 2024-11-15 by NewPaulWalker