OpenXiangShan/XiangShan #3877: `mcause` is different between xiangshan and spike when execute `lh`.

ha0lyu, 2024-11-16:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

When we execute `lh      t6, 979(s9) `, the `mcause` is different between Xiangshan & spike-diff.
```
privilegeMode: 3
 mcause different at pc = 0x0010000008, right= 0x0000000000000005, wrong = 0x0000000000000004
Core 0: [31mABORT at pc = 0x9396629aae80
```

And NEMU got this:
```
privilegeMode: 3
Core 0 dump: DUT critical_error diff REF 
Core 0: [31mABORT at pc = 0xe
```

### Expected behavior

XiangShan supports misaligned address access, so it should trap "load address fault", which is 5.

### To Reproduce

```
lh      t6, 979(s9) 
```

### Environment

- XiangShan branch: master
- XiangShan commit id: 011f1effd
- NEMU commit id: b9507fbc
- SPIKE commit id:

### Additional context

I have searched the previous issues and found some relevant.

ha0lyu, 2024-11-16:
Sorry, I forgot to add the log files: 
[mcause-nemu.log](https://github.com/user-attachments/files/17783334/mcause-nemu.log)
[mcause-spike.log](https://github.com/user-attachments/files/17783335/mcause-spike.log)

NewPaulWalker, 2024-11-18:
Duplicate with https://github.com/OpenXiangShan/XiangShan/issues/3878.
See https://github.com/OpenXiangShan/XiangShan/issues/3878#issuecomment-2481913023 for details.

closed 2024-11-18 by ha0lyu