OpenXiangShan/XiangShan #3878: `mcause` is different between xiangshan and spike when execute `sh`.

ha0lyu, 2024-11-16:
Hi @lewislzh,
I really appreciate what you have done.
I have tested it in **the newest version**, it works! No more `mstatus` different for now.
However, this test case raise `mcause` different:
```
sh      s2, 1259(s0)   # test case
-------------------
privilegeMode: 3
 mcause different at pc = 0x0080000030, right= 0x0000000000000007, wrong = 0x0000000000000006
```
I ran this test case, the `mcause` value is 6 in the official spike, which is the same as XiangShan. But in spike-diff, it is 7, nemu-diff reported nothing.

Thanks.

_Originally posted by @ha0lyu in https://github.com/OpenXiangShan/XiangShan/issues/3860#issuecomment-2476616185_

NewPaulWalker, 2024-11-18:
XiangShan and NEMU consider the range 0~0x8000_0000 as MMIO space, but currently, we do not support unaligned memory access for MMIO. Therefore, raising an address misalign exception is expected behavior.
However, whether unaligned access is supported in Spike depends on a global setting. In our setup, unaligned access is enabled in Spike, so during differential testing with Spike, it raises an access fault instead of an address misalign exception. 

Additionally, you mentioned that the official Spike reports an `mcause` of 6. This is likely because your configured Spike does not support unaligned memory access.

ha0lyu, 2025-04-13:
Hi @NewPaulWalker,
When execute `sd` instruction when `s6 = 0x0, t2 = 0x0`, a mismatch happened when the reference model is `./ready-to-run/riscv64-spike-so`:

[image]
[image]

When the reference model is `./ready-to-run/riscv64-nemu-interpreter-so`, the `mcause = 6`.
**Is this issue related to XiangShan's lack of support for unaligned access in MMIO?**

Test code:
```
_start:
        sd      s6, 4(t2)
        j _start
```
Xiangshan commit id: 724e3eb41
ready-to-run commit id: fb2a729

NewPaulWalker, 2025-04-14:
> Is this issue related to XiangShan's lack of support for unaligned access in MMIO?

Yes! I believe so — this diff is expected behavior when XiangShan does not support unaligned memory accesses to MMIO space.

NewPaulWalker, 2025-04-17:
> Using NEMU as the reference model, its behavior is consistent with XiangShan.
> I suppose this is an issue that exists for the sake of alignment with XiangShan, and it can't be fixed for now, right?

NEMU is indeed aligned with XiangShan, but this is actually not something that needs to be fixed, since we probably don’t plan to support unaligned MMIO accesses for quite a long time.

NewPaulWalker, 2025-04-17:
> I suppose this is an issue that exists for the sake of alignment with XiangShan, and it can't be fixed for now, right?

Additionally, NEMU supports configuring whether to check for unaligned MMIO accesses. In the `configs` directory of NEMU, we use the `riscv64-xs-ref_defconfig` and `riscv64-xs-diff-spike_defconfig` files. In these files, we configure `CONFIG_MMIO_AC_SOFT`.  

For the golden model used in differential testing with XiangShan (`riscv64-xs-ref_defconfig`), unaligned MMIO accesses need to be checked.  
However, for the DUT used in differential testing between NEMU and Spike (`riscv64-xs-diff-spike_defconfig`), unaligned MMIO accesses are no longer checked.

ha0lyu, 2025-04-17:
> NEMU is indeed aligned with XiangShan, but this is actually not something that needs to be fixed, since we probably don’t plan to support unaligned MMIO accesses for quite a long time.

I think the problem is about [spike-so](https://github.com/OpenXiangShan/riscv-isa-sim). This reference model is not aligned with XiangShan. It makes a lot of difference when we test XiangShan with it.

NewPaulWalker, 2025-04-18:
> I think the problem is about [spike-so](https://github.com/OpenXiangShan/riscv-isa-sim). This reference model is not aligned with XiangShan. It makes a lot of difference when we test XiangShan with it.

We’re not currently focusing on this area, so this task has a lower priority. However, if you’re interested, you’re welcome to contribute code to implement this feature consistent with XiangShan.