OpenXiangShan/XiangShan #3882: The initial value of `mnscratch` is inconsistent

ha0lyu, 2024-11-16:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

The initial value of `mnscratch` is `0xcc2c5fc23493516d` in XiangShan, it is 0x0000 in NEMU.

[image]
[image]

### Expected behavior

The initial value of `mnscratch` is consistent.

### To Reproduce

```
    .word 0x74102e73  #csrr t*, mn*
    .word 0x74102273
    .word 0x744022f3
    .word 0x74002473

end:    
    li a0, 0  
    .word 0x5006b
```

### Environment

- XiangShan branch: master
- XiangShan commit id: 011f1effd
- NEMU commit id: b9507fbc
- SPIKE commit id:

ha0lyu, 2024-11-16:
Same as `htimedelta`:
```
 .word 0x60502e73     # csrr    t3, htimedelta
```

ha0lyu, 2024-11-17:
Hi @fly-1011, 
I did not check NEMU issue lastnight, it seems that we find this problem by different methods. Actually, there are lots of test cases triggered this different, and they are same value:
[image]
_Part of the statistic result: 6/n of results when mnstratch is different._

It is indeed strange that we got different `mnscratch` value.

lewislzh, 2024-11-18:
@ha0lyu @fly-1011,
Thank you for your issue. Since the `mnscratch`, `htimedelta`, `tdata2`, and other CSR registers do not have default values specified in the manual, it is reasonable for XiangShan and references (NEMU/Spike) to assign arbitrary initial values. We believe that for RW-type CSR registers without default values specified in the manual, software should not perform read operations before writing to them. Therefore, you may need to modify your test cases accordingly.

ha0lyu, 2024-11-18:
Hi @lewislzh,
Thanks for your reply. 
From the perspective of chip design, I agree with you that when the manual does not specify the initial value of CSR registers, there is no need to assign them. However, from the perspective of security, I think we **can not reply on software to initial CSR registers** before read it and write it. Because software is not reliable as CPU, you believe "_software should not perform read operations before writing to them_", but you **can not guarantee it will**. 
Software is prone to errors and can be easily hacked, but hardware is not. Take Intel CPU as an example: Intel released a security  [advisory](https://www.intel.com/content/www/us/en/security-center/advisory/intel-sa-00929.html) in January, the [CVSS is high](https://nvd.nist.gov/vuln/detail/CVE-2023-32282). I don't know the detail of this vulnerability, but as we can see, the BIOS is not reliable. This is [another advisory](https://www.intel.com/content/www/us/en/security-center/advisory/intel-sa-00527.html?s=09) containing **16 bugs related to BIOS**. There are lots of security vulnerabilities related to BIOS or UEFI firmware in Intel, AMD and ARM CPUs. **Can we still rely on software to guarantee security?** **The best way to guarantee the security and privacy of users is in hardware layer, not software.** Of course, this is not a bug of XiangShan, but is a potential security risk for XiangShan users.

Thanks.

lewislzh, 2024-11-18:
@ha0lyu 
The behavior of such CSR registers having random initial values is valid in both XiangShan and NEMU. Therefore, using difftest to compare initial values in this case is meaningless since these values are random. 
We recommend that tests avoid directly reading CSR registers with unknown initial values when running in the difftest framework. We do not consider such XiangShan and NEMU difftest mismatches as hardware issues.

ha0lyu, 2024-11-18:
Thanks for your explanation and suggestion. 

> The behavior of such CSR registers having random initial values is valid in both XiangShan and NEMU. Therefore, Using difftest to compare initial values is meaningless because these values are random, and you cannot compare whether two random values are consistent.

1. I did not know it is random value. I have [tens of test case](https://github.com/OpenXiangShan/XiangShan/issues/3882#issuecomment-2481316244) show that the initial value is the same one.
2. Yes, when they are random value, there is no meaning to compare them. And I will avoid reading those random initial values and compare them.

Thanks.

closed 2024-11-18 by ha0lyu

lewislzh, 2024-11-18:
This random value is determined at compile time, so if you haven't recompiled XiangShan, the values produced by the same emulator will indeed be the same.