OpenXiangShan/XiangShan #3906: make verilog Error [Makefile:154: build/rtl/XSTop.sv] Error 1

quanchenliu, 2024-11-20:
### Before start

- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

I followed the [Xiangshan documentation](https://docs.xiangshan.cc/zh-cn/latest/tools/xsenv/#tldr) for deployment. However, the following error occurred when running `make verilog`:
> lqc@lqc-virtual-machine:~/xs-env/XiangShan$ make verilog
mkdir -p build/rtl
time -avp -o ./build/time.log mill -i xiangshan.runMain top.TopMain   \
	--target-dir build/rtl --config DefaultConfig --issue B --firtool-opt "--repl-seq-mem --repl-seq-mem-file=XSTop.sv.conf"		\
	--num-cores 1 --dump-fir --target systemverilog --split-verilog --firtool-opt "-O=release --disable-annotation-unknown --lowering-options=explicitBitcast,disallowLocalVariables,disallowPortDeclSharing,locationInfoStyle=none" --fpga-platform --disable-all --remove-assert --reset-gen --disable-always-basic-diff
[build.sc] [54/59] compile 
[info] compiling 8 Scala sources to /home/lqc/xs-env/XiangShan/out/mill-build/compile.dest/classes ...
[info] done compiling
[157/557] rocketchip.macros.compile 
[info] compiling 1 Scala source to /home/lqc/xs-env/XiangShan/out/rocketchip/macros/compile.dest/classes ...
[info] done compiling
[189/557] rocketchip.hardfloat.compile 
[info] compiling 20 Scala sources to /home/lqc/xs-env/XiangShan/out/rocketchip/hardfloat/compile.dest/classes ...
[info] done compiling
[215/557] rocketchip.cde.compile 
[info] compiling 1 Scala source to /home/lqc/xs-env/XiangShan/out/rocketchip/cde/compile.dest/classes ...
[info] done compiling
[244/557] rocketchip.compile 
[info] compiling 316 Scala sources to /home/lqc/xs-env/XiangShan/out/rocketchip/compile.dest/classes ...
[warn] 8 deprecations
[warn] 67 deprecations (since 2.13.0)
[warn] 1 deprecation (since 2.13.7)
[warn] 1 deprecation (since 9)
[warn] 52 deprecations (since CDE 0.1)
[warn] 1 deprecation (since rocket-chip 1.2)
[warn] 130 deprecations in total; re-run with -deprecation for details
[warn] 19 feature warnings; re-run with -feature for details
[warn] 8 warnings found
[info] done compiling
[288/557] difftest.compile 
[info] compiling 18 Scala sources to /home/lqc/xs-env/XiangShan/out/difftest/compile.dest/classes ...
[warn] 7 deprecations
[warn] 1 deprecation (since 2.13.0)
[warn] 2 deprecations (since Chisel 6.0)
[warn] 10 deprecations in total; re-run with -deprecation for details
[warn] four warnings found
[info] done compiling
[323/557] utility.compile 
[info] compiling 40 Scala sources to /home/lqc/xs-env/XiangShan/out/utility/compile.dest/classes ...
[warn] /home/lqc/xs-env/XiangShan/utility/src/main/scala/utility/PerfCounterUtils.scala:35:26: a pure expression does nothing in statement position
[warn]       case regular(_) => true
[warn]                          ^
[warn] /home/lqc/xs-env/XiangShan/utility/src/main/scala/utility/TLUtils/TLLogger.scala:91:5: match may not be exhaustive.
[warn] It would fail on the following input: TLBundleE()
[warn]     chn match {
[warn]     ^
[warn] 2 deprecations (since 2.13.0); re-run with -deprecation for details
[warn] three warnings found
[info] done compiling
[348/557] huancun.compile 
[info] compiling 44 Scala sources to /home/lqc/xs-env/XiangShan/out/huancun/compile.dest/classes ...
[warn] /home/lqc/xs-env/XiangShan/huancun/src/main/scala/huancun/Slice.scala:442:44: non-variable type argument huancun.noninclusive.MSHR in type pattern Seq[huancun.noninclusive.MSHR] (the underlying of Seq[huancun.noninclusive.MSHR]) is unchecked since it is eliminated by erasure
[warn]     case (dir: noninclusive.Directory, ms: Seq[noninclusive.MSHR]) =>
[warn]                                            ^
[warn] /home/lqc/xs-env/XiangShan/huancun/src/main/scala/huancun/Slice.scala:457:38: non-variable type argument huancun.inclusive.MSHR in type pattern Seq[huancun.inclusive.MSHR] (the underlying of Seq[huancun.inclusive.MSHR]) is unchecked since it is eliminated by erasure
[warn]     case (_: inclusive.Directory, _: Seq[inclusive.MSHR]) =>
[warn]                                      ^
[warn] 1 deprecation
[warn] 6 deprecations (since 2.13.0)
[warn] 7 deprecations in total; re-run with -deprecation for details
[warn] 3 feature warnings; re-run with -feature for details
[warn] 6 warnings found
[info] done compiling
[384/557] coupledL2.compile 
[info] compiling 58 Scala sources to /home/lqc/xs-env/XiangShan/out/coupledL2/compile.dest/classes ...
[warn] /home/lqc/xs-env/XiangShan/coupledL2/src/main/scala/coupledL2/tl2chi/TL2CHICoupledL2.scala:127:20: non-variable type argument coupledL2.tl2chi.Slice in type pattern Seq[coupledL2.tl2chi.Slice] (the underlying of Seq[coupledL2.tl2chi.Slice]) is unchecked since it is eliminated by erasure
[warn]       case slices: Seq[Slice] =>
[warn]                    ^
[warn] 1 deprecation
[warn] 2 deprecations (since 2.13.0)
[warn] 3 deprecations in total; re-run with -deprecation for details
[warn] four warnings found
[info] done compiling
[416/557] openLLC.compile 
[info] compiling 23 Scala sources to /home/lqc/xs-env/XiangShan/out/openLLC/compile.dest/classes ...
[warn] 3 deprecations (since 2.13.0); re-run with -deprecation for details
[warn] one warning found
[info] done compiling
[448/557] yunsuan.compile 
[info] compiling 74 Scala sources to /home/lqc/xs-env/XiangShan/out/yunsuan/compile.dest/classes ...
[info] done compiling
[480/557] fudian.compile 
[info] compiling 15 Scala sources to /home/lqc/xs-env/XiangShan/out/fudian/compile.dest/classes ...
[warn] /home/lqc/xs-env/XiangShan/fudian/src/main/scala/fudian/FPToInt.scala:55:12: a pure expression does nothing in statement position; multiline expressions may require enclosing parentheses
[warn]   io.to_s2.sel_lpath
[warn]            ^
[warn] 5 deprecations (since 2.13.0); re-run with -deprecation for details
[warn] two warnings found
[info] done compiling
[510/557] macros.compile 
[info] compiling 1 Scala source to /home/lqc/xs-env/XiangShan/out/macros/compile.dest/classes ...
[info] done compiling
[547/557] xiangshan.compile 
[info] compiling 364 Scala sources to /home/lqc/xs-env/XiangShan/out/xiangshan/compile.dest/classes ...
[info] done compiling
[557/557] xiangshan.runMain 
FPGASoC cores: 1 banks: 4 block size: 64 bus size: 256
......
... Too much content to display here....
......
[warn] There were 127 warning(s) during hardware elaboration.
1 targets failed
xiangshan.runMain Subprocess failed
make: *** [Makefile:154: build/rtl/XSTop.sv] Error 1

### What did you do before

Firstly, I have expanded the Swap file before.  I present the memory information and the contents of the Swap file:
>$ free -m
                      total        used        free      shared  buff/cache   available
Mem:            7894         999        6363          48         532        6589
Swap:          32767        1085       31682

Secondly, my disk space allocation is as follows:
>$ df -h
Filesystem      Size  Used Avail Use% Mounted on
tmpfs           790M  2.1M  788M   1% /run
/dev/sda3       147G   55G   86G  39% /
tmpfs           3.9G     0  3.9G   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
/dev/sda2       512M  6.1M  506M   2% /boot/efi
tmpfs           790M  120K  790M   1% /run/user/1000
/dev/sr0        4.5G  4.5G     0 100% /media/lqc/Ubuntu 22.04.5 LTS amd64

Thirdly, I am not sure whether `make init` has fully cloned the submodules, so I provide the corresponding output below:

> $ make init
git submodule update --init
cd rocket-chip && git submodule update --init cde hardfloat

### Environment

- Operation System: Ubuntu-22.04-LTS
- XiangShan branch: 
- XiangShan commit id: 
- NEMU commit id: 
- SPIKE commit id: 
- Operating System: 
- gcc version: 11.4.0
- mill version: 0.11.8
- java version: openjdk 11.0.25

### Additional context

I am a greenhand who has just started exploring the RISC-V field, so please forgive the simplicity of my questions. 
In previous inquiries, I couldn't find a solution related to `[Makefile:154]` errors. However, I did come across issues related to `[Makefile:153]` and `[Makefile:155]`, but I am not sure if they are relevant.  
I sincerely hope you can guide me through this.

Ma-YX, 2024-11-20:
The log information you provided is incomplete, and it is not possible to determine the specific cause of your issue. You can choose to output the log to a TXT file and upload the file as an attachment.

About memory requirements, you can refer to https://github.com/OpenXiangShan/XiangShan/issues/2451.
Two methods:
* Increase the system's swap space so that the total memory + swap reaches over 64 GB.
* Compile with CIRCT. You can try adding the MFC=1 parameter during make emu.

Regarding the "whether make init correctly" issue, a simple approach is to directly enter the submodule (e.g., coupledL2/yunsuan) and check if it is empty (or check its status).

quanchenliu, 2024-11-21:
First, I have expanded the swap file to 64GB.
> $ free -h
                    total        used        free      shared  buff/cache   available
Mem:           7.7Gi       1.3Gi       4.7Gi        66Mi       1.7Gi       6.1Gi
Swap:           63Gi          0B        63Gi

I executed `make clean` and then reinitialized. I found `XiangShan/coupledL2` , but I did not find the `yunsuan` file in that folder.
[image]

To help you better understand the issue I encountered, I executed `make verilog 2>&1 | tee output.txt` and uploaded the `output.txt` file. [output.txt](https://github.com/user-attachments/files/17839512/output.txt)

quanchenliu, 2024-11-21:
> The log information you provided is incomplete, and it is not possible to determine the specific cause of your issue. You can choose to output the log to a TXT file and upload the file as an [...]

First, I have **expanded the swap file to 64GB**.

>$ free -h
> total used free shared buff/cache available
> Mem: 7.7Gi 1.3Gi 4.7Gi 66Mi 1.7Gi 6.1Gi
> Swap: 63Gi 0B 63Gi

I executed make clean and then reinitialized. I am sure that `XiangShan/yunsuan`  or `XiangShan/coupledL2`is **not empty**.

>~/xs-env/XiangShan$ lm ./yunsuan
total 44
-rw-rw-r-- 1 lqc lqc  697 11月 21 10:32 build.sbt
-rw-rw-r-- 1 lqc lqc 2063 11月 21 10:32 build.sc
-rw-rw-r-- 1 lqc lqc 9617 11月 21 10:32 LICENSE
-rw-rw-r-- 1 lqc lqc 1936 11月 21 10:32 Makefile
-rw-rw-r-- 1 lqc lqc  674 11月 21 10:32 Makefile.softfloat
drwxrwxr-x 2 lqc lqc 4096 11月 21 10:32 project
-rw-rw-r-- 1 lqc lqc  665 11月 21 10:32 README.md
drwxrwxr-x 2 lqc lqc 4096 11月 21 10:32 scripts
drwxrwxr-x 4 lqc lqc 4096 11月 21 10:32 src
~/xs-env/XiangShan$ lm ./coupledL2
total 60
-rw-rw-r-- 1 lqc lqc  2889 11月 21 10:32 build.sc
-rw-rw-r-- 1 lqc lqc   277 11月 21 10:32 common.sc
drwxrwxr-x 2 lqc lqc  4096 11月 21 10:32 HuanCun
-rw-rw-r-- 1 lqc lqc  9718 11月 21 10:32 LICENSE
-rw-rw-r-- 1 lqc lqc 11348 11月 21 10:32 LICENSE.SiFive
-rw-rw-r-- 1 lqc lqc  1914 11月 21 10:32 Makefile
-rw-rw-r-- 1 lqc lqc   160 11月 21 10:32 README.md
drwxrwxr-x 2 lqc lqc  4096 11月 21 10:32 rocket-chip
drwxrwxr-x 2 lqc lqc  4096 11月 21 10:32 scripts
drwxrwxr-x 4 lqc lqc  4096 11月 21 10:32 src
drwxrwxr-x 2 lqc lqc  4096 11月 21 10:32 utility

Next, I executed the command `make verilog 2>&1 | tee output.txt`, but it still failed. To help you understand the issue I encountered, I have uploaded the output file [output.txt](https://github.com/user-attachments/files/17839649/output.txt).

eastonman, 2024-11-21:
There is still nothing useful we can get from you log.

These lines below show the compiling process is running normally.
```
[BackendParams] wbType: INT(-1), rdType: IntRD(-1,2147483647)
[BackendParams] wb port 0 rdcfgs: List(IntRD(0,0), IntRD(1,0), IntRD(6,1), IntRD(7,1))
[BackendParams] wb port 5 rdcfgs: List(IntRD(8,0))
[BackendParams] wb port 1 rdcfgs: List(IntRD(2,0), IntRD(3,0), IntRD(4,1), IntRD(5,1))
[BackendParams] wb port 6 rdcfgs: List(IntRD(9,0))
...
```
Then it cames to a sudden stop. To me this looks like an OOM-kill or something similar.

I suggest monitoring memory usage during the compile process. Or just use `dmesg` to check whether a OOM-kill did occur.

quanchenliu, 2024-11-21:
1. I followed your advice and used the `top` command to monitor CPU and Swap file usage during the execution of `make verilog`, and did not observe any memory overflow. 
2. After the `make verilog` execution failed, I ran `sudo dmesg | grep -i oom`, but it produced no output.. 
3. I uploaded the latest output to the [ouput2.txt](https://github.com/user-attachments/files/17844615/ouput2.txt)  file. 
4. I compared `output.txt` and `output2.txt` files locally using `diff`, and found that the content after line 324 has changed.

I hope the information above provides a clearer picture of the issue I'm facing. I appreciate your assistance and look forward to your help!

Ma-YX, 2024-11-21:
According to your latest output (output2.txt), I think you have already executed `make verilog` successfully: (line 4076)
```
(List(int),IntRange(0,2))                                                          
(List(int),IntRange(0,1))                                                          
(List(int),IntRange(0,1))                                                          
(List(int),IntRange(0,1))                                                          
./scripts/gen_sep_mem.sh "./scripts/vlsi_mem_gen" "build/rtl/XSTop.sv.conf" "build/rtl"
```
You can enter `$(YOUR_PATH)/XiangShan/build/rtl` to check whether there are verilog files.

quanchenliu, 2024-11-21:
> According to your latest output (output2.txt), I think it's possible that you have already executed `make verilog` successfully: (line 4076)
> 
> ```
> (List(int),IntRange(0,2))                                                          
> (List(int),IntRange(0,1))                                                          
> (List(int),IntRange(0,1))                                                          
> (List(int),IntRange(0,1))                                                          
> ./scripts/gen_sep_mem.sh "./scripts/vlsi_mem_gen" "build/rtl/XSTop.sv.conf" "build/rtl"
> ```
> 
> You can enter `$(YOUR_PATH)/XiangShan/build/rtl` to check whether there are verilog files.

Sorry, I used the `grep` command in the `xs-env` directory and could not find the `XStop.v` file, so I assumed the generation failed and delete the `xs-env` to restart this work.

1. Environment：
   ```
   Mem: 8G		Swap: 64G	Disk:70G
   gcc --version:			gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
   java --version:			openjdk 11.0.25 2024-10-15
   mill --version:			Mill Build Tool version 0.11.8
   verilator --version:	Verilator 5.028 2024-08-21 rev v5.028
   ```
2. I executed `make init` with the device having normal access to github.com. I confirm that the files `XiangShan/yunsuan` and `XiangShan/coupledL2` are **non-empty**.

3. [Troubleshooting ](https://docs.xiangshan.cc/zh-cn/latest/tools/troubleshoot-zh/) recommends running `make emu CONFIG=MinimalConfig` to verify that the XiangShan environment is configured correctly.

4. I followed your advice and used the **top** command to monitor CPU and Swap file usage during the execution of make verilo[Thg, and **did not observe any memory overflow**. Also, after the `make emu` , I ran `sudo dmesg | grep -i oom`, and it **produced no output**.

5. I uploaded the latest output to the [output_Minimal.txt](https://github.com/user-attachments/files/17848535/output_Minimal.txt) file.

I sincerely apologize if my impulsive actions have caused any inconvenience to your work. I appreciate your assistance and look forward to your help!

Ma-YX, 2024-11-22:
There's no need to feel troubled about asking questions; we'll do our best to address the issues in the thread. 

I checked the log and didn't find any errors—it seems the `make emu CONFIG=MinimalConfig` process completed successfully. However, based on the information you provided, the file you should be looking for is **`XSTop.sv`** rather than `XSTop.v`.

The `make verilog` command in current XiangShan generates `SystemVerilog` files. In the previous message, I was merely pointing out that checking for the corresponding Verilog file might have caused you some confusion—my apologies for that.

quanchenliu, 2024-11-22:
Yes! I find `XStop.sv` in the direcory `XiangShan/build/rtl`. Your answer is a good news for me！
Thanks for your help sincerely! Otherwise, maybe we should give the [XiangShan Documents](https://docs.xiangshan.cc/zh-cn/latest/tools/xsenv-en/#generate-verilog-code-for-xiangshan-core) a little update... 
you know, just in case!

closed 2024-11-22 by quanchenliu