OpenXiangShan/XiangShan #3926: Bug when using `amoand.w` to change instructions in memory.

ha0lyu, 2024-11-25:
### Before start

- [X] I have read the [RISC-V ISA Manual](https://github.com/riscv/riscv-isa-manual) and this is not a RISC-V ISA question. 我已经阅读过 RISC-V 指令集手册，这不是一个指令集本身的问题。
- [X] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [X] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [X] I have reviewed the commit messages from the relevant commit history. 我已经浏览过相关的提交历史和提交信息。

### Describe the bug

Before run `amoand.w`, a handler function has been set, and `(ra)` holds the first instruction of the handler function. Then, run the test function:
```
    .word 0x6620a1af    # amoand.w gp, sp, (ra)
    mv a5, gp       # a5 is original (ra) value, same as gp
    lw t2, 0(ra)    # t2 is the (ra) value now, which is zero.
    mv t0, zero     # t0 = 0x0.
    .word 0xffffff      # trap EX_II, jump to handler function.
```
Because of `sp=0x00000000`, `(ra)` will be zero. When jumping to the handler function, the first instruction should be 0x00000000. 
If we run this test case in XiangShan, we will get:
[image]
[image]

### Expected behavior

Look the beaver in spike:
[image]

### To Reproduce

[amoand.zip](https://github.com/user-attachments/files/17899948/amoand.zip)

### Environment

- XiangShan branch: master
- XiangShan commit id: 3dfb64bde
- NEMU commit id: b2fb5272
- SPIKE commit id:

NewPaulWalker, 2024-11-25:
@ha0lyu !
If you want to modify program itself, you should use the `fence.i` instruction to ensure that instruction fetching gets the latest memory data.

lewislzh, 2024-11-25:
Also, although this test can hit a good trap after adding `fence.i`, it still triggers a `critical-error`. It is unclear whether this is the intended result of the test. We recommend checking whether there are any critical-error logs even after hitting a good trap. Such behavior involving multiple traps is likely a test issue.

ha0lyu, 2024-11-25:
> Also, although this test can hit a good trap after adding `fence.i`, it still triggers a `critical-error`. It is unclear whether this is the intended result of the test. We recommend checking [...]

This test case is a part of another test case, which triggered the DiffTest to report difference between XiangShan & NEMU. I think the **intended result** is the consistence between XiangShan & NEMU.

lewislzh, 2024-11-25:
> > Also, although this test can hit a good trap after adding `fence.i`, it still triggers a `critical-error`. It is unclear whether this is the intended result of the test. We recommend checking whether there are any critical-error logs even after hitting a good trap. Such behavior involving multiple traps is likely a test issue.
> 
> This test case is a part of another test case, which triggered the DiffTest to report difference between XiangShan & NEMU. The **intended result** is the consistence between XiangShan & NEMU.

Got it, I just wanted to remind you not to overlook the `critical-error` warning, even if it is a Good Trap.

lewislzh, 2024-11-25:
> Yes, when add `fence.i`, it triggers `critical-error`. It seems that not all differences reported by DiffTest are bugs. Right?
Since the NEMU simulator lacks certain hardware concepts, there may be cases where RTL and the simulator differ, but these are not bugs. This primarily involves caches and page tables. The issue you reported occurs because NEMU does not simulate hardware structures related to the instruction cache, resulting in the absence of the concept of consistency. This allows NEMU to immediately read the modified instructions after self-modifying code, even before a fence, whereas RTL does not permit this.

The incomplete simulation of caches and page tables is partly due to the complexity of simulating features like replacement policies and partly because NEMU is an instruction set simulator rather than a microarchitecture simulator, so it does not need to simulate caches or page tables. In such cases, software may need to make some adaptations, such as adding fence instructions. However, in most cases, when a diff error occurs, it indicates a bug.

ha0lyu, 2024-11-25:
Get it, thank you very much.

closed 2024-11-25 by ha0lyu